modelica/ModelicaStandardLibrary #3055: update Labels

AHaumer, 2019-07-17:
@beutlich Shouldn't we update Labels at GitHub?
Electrical.QuasiStationary -> Electrical.QuasiStatic
Electrical.MultiPhase -> Electrical.PolyPhase
I know the old terms are used in 3,2,3 but shouldn't we switch to the new terms?
I'm pretty sure I forgot some Labels ...
Furthermore, wouldn't it be usefull to have "top level" Labels like "Blocks", "Electrical", ...

AHaumer, 2019-07-17:
Maybe for some time we use 
Electrical.QuasiStatic/QuasiStationary
Electrical.PolyPhase/MultiPhase

beutlich, 2019-07-17:
I already thought about it. The issue is that I generate the ResolvedGitHubIssues.html from the closed issues and sort by labels. If I do it for MSL v3.2.3+build.3 I still need the old labels. Thus I have no real idea and left it as is.

AHaumer, 2019-07-17:
Is using old and new labels in prallel the solution?
Old labels -> 3.2.3, new labels -> >4.0.0

beutlich, 2019-07-17:
That would work out (and keeps me busy with relabeling).

beutlich, 2019-07-17:
Done. Added new labels and relabeled all open and closed issues/PRs w.r.t. the new labels.

closed 2019-07-17 by beutlich