modelica/ModelicaStandardLibrary #3057: Coding / Naming guidelines

AHaumer, 2019-07-17:
There's a lot of discussion about coding and naming guidelines. 
Just a question:
How to represent a variable from a textbook with an underscore as a Modelica variable?
Widely used for Complex variables.
[image]
How to represent a variable with a line as accent ("upperscore", I don't know the right term) as a Modelica variable?
Widely used for the arithmetic mean of a variable or as the negated variable.
[image]
@beutlich @dietmarw @christiankral @tobolar ... - any idea?

christiankral, 2019-07-17:
This topic is related with https://github.com/modelica/ModelicaSpecification/issues/2357

beutlich, 2019-07-17:
I think the issue is not feasible for MSL v4.0.0 at all.

I also think that modelica/ModelicaSpecification#2357 is the right space to discuss it together with the Wolfram guys (@henrikt-ma, @maltelenz) who do it right in Mathematica.

AHaumer, 2019-07-18:
appended to modelica/ModelicaSpecification#2357

closed 2019-07-18 by AHaumer