modelica/ModelicaStandardLibrary #3060: Code / Formatting guidelines checker

AHaumer, 2019-07-18:
In a discussion the idea came up to have tool independent checker for coding and formatting. 
I know that some parts of such a thing exist in some tools, but it would be nice to have a tool independent solution (like the regression check!). 
Let the tool run through all models of a library, getting a list of things tho improve, like:
parameter description missing, icon color is wrong, name in icon layer has wrong size and color
and many more - fortunately we have guidelines in the UsersGuide.

beutlich, 2019-10-09:
We can discuss it at the scheduled MAP-Lib strategy meeting in Nov 2019.

atrosinenko, 2020-01-07:
I have [recently implemented](https://github.com/pmd/pmd/pull/2041) an initial support of Modelica in [PMD](https://github.com/pmd/pmd/) static code analyzer. For now, it is _experimental_ (but already merged into `master`) and has only demo rules, but:
* PMD has language-agnostic "XPath rule" built-in, so you can write some XPath queries that mark matched nodes as violations - without any Java coding at all, just external XMLs with rules
  * one of these demo rules is

        //ClassSpecifier/*[SimpleName[1]/@Image != SimpleName[last()]/@Image]

    it basically checks for `model Name end OtherName;` mistakes
* PMD Modelica language module already has some more-or-less MLS-aligned flattening process to resolve type references
    * the other demo rule tries to find suspicious class name shadowing
    * requires Java coding to utilize
* PMD has Copy-Paste Detector module that can find large-enough copied parts
  * in some sense, it is even harder to review: if someone sends a bad pull request, it would probably be rejected or improved by maintainers, but if some other PR has parts of code copied from other places of the same repo (already reviewed and good quality!), then maintainer need to know **all codebase at once** to notice copy-paste
  * it seems, it already found a [180-line](https://github.com/pmd/pmd/pull/2041#issuecomment-539368857) copy-paste in MSL

It is still not published as part of the released Binary Distribution (the next release is planned at the end of January), so manual build of `master` branch is required. Also, it is still experimental, so some API still may change.

Last but not least: sometimes Modelica compilers crash with `Internal compiler error` or something like this. There exist many different tools to minimize source code that crash the compiler. I have implemented yet another as a tool for PMD that should support any "fully-featured" (that is, not CPD-only) PMD language (including Modelica). Its maintainers placed it to separate repo [pmd/pmd-scm](https://github.com/pmd/pmd-scm) as another experimental feature.

PS: On "coding and formattig": generally, PMD cannot check formatting, since it works with AST. On the other hand, it can trivially check some assertions about coding style such as "every package name starts with `p` letter".

beutlich, 2020-01-08:
@atrosinenko This is very good news and I will give PMD a try once the new version is available. (I already watched the PMD releases, so I will get notified when it is released.

I also addressed the [180-line](https://github.com/pmd/pmd/pull/2041#issuecomment-539368857) copy&paste by #3296. Thanks again for reporting.

atrosinenko, 2020-01-09:
I have published a [preview build](https://github.com/atrosinenko/pmd/releases/tag/modelica-preview) in my fork. But it is quite simple to build it on Linux (at least, on a fresh Ubuntu) provided you have OpenJDK installed: you just have to clone [pmd/pmd](https://github.com/pmd/pmd) and run `./mvnw clean verify` - you will get `pmd-bin-<...>.zip` in the `pmd-dist/target` directory.

It may be worth testing it before the official release because now it has very few rules (basically, these are more demos) and it would be great if some actual rules be implemented (or just suggested) by real Modelica developers.

atrosinenko, 2020-02-11:
Modelica support for PMD was released as part of version 6.21.0. Meanwhile, it may be worth adding to https://modelica.org/tools (with the `experimantal` note).

beutlich, 2020-02-11:
> 
> 
> Modelica support for PMD was released as part of version 6.21.0. Meanwhile, it may be worth adding to https://modelica.org/tools (with the `experimantal` note).

@sjoelund Can you please add PMD to https://modelica.org/tools as suggested. Thanks.

sjoelund, 2020-02-12:
Pinging @adeas31 who does most of the webmaster tasks

beutlich, 2020-05-08:
> 
> 
> Pinging @adeas31 who does most of the webmaster tasks

Did not happen so far. See also https://github.com/modelica/BackOffice/issues/35.

adeas31, 2020-05-19:
Added PMD to the tools page.