modelica/ModelicaStandardLibrary #3064: Singular inconsistent scalar system for saturatingInductor

AHaumer, 2019-07-20:
I found the above mentioned bug, when current through the saturatingInductor goes rather smoothly and slowly through zero - see enclosed example.
[TestSaturatingInductor.zip](https://github.com/modelica/ModelicaStandardLibrary/files/3413347/TestSaturatingInductor.zip)
Investigating the problem, I don't think this is a tool issue, it's a problem of the implemenation. Well I have to admit I implemented this model back in 2004. Honestly I don't know since which release we're living with that bug - in many cases this bug won't appear. However, MSL should be "clean" in as many cases as possible.
I had to dig into modeling of saturation but I found a solution. After discussion and decision with @christiankral about details of the implementation I'll provide a bug fix for maint\3.2.3 and for 4.0.0-dev in 2 PRs before my vacation (during August).

beutlich, 2019-07-20:
> I don't think this is a tool issue, it's a problem of the implemenation.

The TestSaturatingInductor runs nicely in SimulationX 4.0.2 with both DAE or ODE solver.

[image]

beutlich, 2019-07-20:
> I'll provide a bug fix for maint\3.2.3

Since we do not have a detailed git branching model, no commits should go to maint\3.2.3 while MSL v3.2.3+build.3 is not yet relased, i.e., no commits before 2019-09-18. You still can create PRs though,

AHaumer, 2019-07-20:
The bugfix should definitely go to the next release ...

PR #3066 (2019-10-20): Cures #3064 SaturatingInductor for 4.0.0-dev
Resolves #3064 SaturatingInductor, added a test example to ModelicaTest
closed 2019-10-20 by beutlich