modelica/ModelicaStandardLibrary #3076: Invalid call to exp in Modelica.Math.Matrices.integralExpT

Daiwen, 2019-07-29:
In `Modelica.Math.Matrices.integralExpT` line 19, there is a call to `exp` that takes two arguments.
The only built-in function `exp` defined in the specification expects only one argument. I think the intent was to reference `Modelica.Math.Matrices.exp`. My understanding of name lookup is that built-in constructs have priority over contextual names.

If I am correct, I suggest we change the original call to `Matrices.exp`.

sjoelund, 2019-07-29:
There is no priority of built-ins being looked up before other names. The following should work in any Modelica tool:
```
model M
  function sin
    input String s;
     output Real r = 1.5;
  algorithm
  end sin;
  Real r = sin("abc");
end M;
```

There are rules for some built-in names, but they are only Real, Integer, String, Boolean as far as I remember.

beutlich, 2019-07-29:
Hm, the following test model runs in Dymola, SimulationX and Wolfram SystemModeler.

```mo
model T3076
  parameter Real A[2,2] = {{1,2},{3,4}};
  parameter Real B[2,1] = {{1},{1}};
  Real phi[2,2];
  Real g[2,1];
  Real g1[2,1];
equation
  (phi, g, g1) = Modelica.Math.Matrices.integralExpT(A, B, 1.1);
end T3076;
```

Daiwen, 2019-07-29:
I see, I didn't double check how `Modelica.Math.exp` was implemented and I thought there was a conflict.

closed 2019-07-29 by Daiwen