modelica/ModelicaStandardLibrary #3089: Future of Modelica.Mechanics.MultiBody.Visualizers.Internal

beutlich, 2019-08-10:
Since the earliest days of Modelica.Mechanics.MultiBody (MSL v2.1, ee9a1a29568c2815a0e568719f25482a50d388c7) the comment description  of Modelica.Mechanics.MultiBody.Visualizers.Internal claims that visualizers willl be improved in the future

https://github.com/modelica/ModelicaStandardLibrary/blob/d5327d08e485d06bb7eab90bd9641c2a50fb782c/Modelica/Mechanics/MultiBody/Visualizers.mo#L2605-L2606

Later, by 91da30f5f0423f3625dde33902c83f76fa5e8bce for MSL v2.2.1 a "temporary hack" was added to Modelica.Mechanics.MultiBody.Visualizers.Internal for display of 3D font

https://github.com/modelica/ModelicaStandardLibrary/blob/d5327d08e485d06bb7eab90bd9641c2a50fb782c/Modelica/Mechanics/MultiBody/Visualizers.mo#L2831-L2835

 * Modelica.Mechanics.MultiBody.Visualizers.Internal.FixedLines is not used in MSL at all and could be moved to ObsoleteModelica4 without any damage.
 * Modelica.Mechanics.MultiBody.Visualizers.Internal.Lines is used in Modelica.Mechanics.MultiBody.World to display the x-, y- and z-axis labels if enableAnimation and animateWorld and axisShowLabels are all true. (In the MSL distributed by SimulationX animateWorld is set to false by defaulr to improve performance and display a nicer default frame for the origin).

IIRC, there has been no attempt to resolve this issue of 3D font visualization in MAP-Lang or MAP-Lib for the last 15 years. I know that SimulationX can display 3D font objects, but this is independently from the Modelica visualizers. I assume that the DLR Vis Library also can display 3D fonts in a better way.

I am therefore in favour of deprecating the axisShowLabels feature of Modelica.Mechanics.MultiBody.World and move Modelica.Mechanics.MultiBody.Visualizers.Internal.Lines also to ObsoleteModelica4.

@HansOlsson @MartinOtter @DagBruck What do you think?

DagBruck, 2019-09-05:
The biggest concern should be any use outside of MSL, which might be affected. However, your proposal to just move it would not (yet) permanently remove any functionality.

beutlich, 2019-09-05:
Thanks for feedback @DagBruck . Indeed moving both types to to ObsoleteModelica4 now, will not lose functionality of 3D font visualization.

@MartinOtter @tobolar What do you think on removing the axisShowLabels parameter from MultiBody?

tobolar, 2019-09-05:
I agree the current hack is odd but it is reasonable to have some axes description for visualization. If it would be moved to ObsoleteModelica4, I guess there shall be at least some idea what to use instead. Otherwise we indicate to remove the element in the future without having a better option.
How is the chance to handle a 3D text visualization within Modelica Langugage Specification?

beutlich, 2019-09-05:
> How is the chance to handle a 3D text visualization within Modelica Langugage Specification?

If I am not wrong, nothing happened in the last 15 years. Thus there simply was not enough pressure to enforce any improvements ofiicially.

This also means (as of now) that the current comment on the Visualizers.Internal is misleading.

HansOlsson, 2019-09-05:
> How is the chance to handle a 3D text visualization within Modelica Langugage Specification?

We would first need to figure out what we need.
Texts in 3D animations could be:

- Texts on the sides of objects in 3D (they can be part of textures).
- Texts with possible depth in 3D that rotate with the rest of the 3D. (As these labels.)
- Textual labels in 3D-space that are always turned towards the viewer (and sometimes the entire label also moves in other ways). (This is more normal for labels in 3d-plots; think annotated street view.)
- Caption completely outside the 3D-space. (Like "Robot").

Since we haven't even considered the options I doubt that we can get agreement on that.

Thus I think that we should at least update the current comment; and the remaining question is whether we should just remove the xyz-objects (and instead have tools add them in smart ways),  have some special axis-object; or just keep as is.

tobolar, 2019-09-05:
Rethinking about this issue, I guess we can split it into two parts:
1. An object for visualization of a coordinate frame incl. axes description. I'm not sure but maybe we just need a visualization object here which can be simply handled by the tools' visualizers (as SimulationX does).
2. A generic 3D animation of a text, enabling far more options.

I suppose this issue is more about 1. If I'm right, we can open a separate issue for 3D text only.

closed 2019-09-06 by beutlich

tobolar, 2019-09-06:
Even if #3111 is fine, I don't believe it really closes this issue as the title is "Future of ...". It actually could only be closed if there is another corresponding issue on 3D text in Modelica Spec.

beutlich, 2019-09-08:
> 
> 
> Even if #3111 is fine, I don't believe it really closes this issue as the title is "Future of ...". It actually could only be closed if there is another corresponding issue on 3D text in Modelica Spec.

From a library point of view I am fine with the current change (disclaimer: was my own PR). Regarding the actual promise of supporting fonts/text in 3D visualization I'd like to ask @TorstenBlochwitz (as MA technology officer) to join the discussion and open a new spec ticket if applicable. Thanks.

PR #3111 (2019-09-06): Remove misleading promise of 3D font visualization
close #3089