modelica/ModelicaStandardLibrary #3094: Heat exchanger example - Heat conduction through the wall

Ciobrelis, 2019-08-15:
Hello, 
I am not sure if it is a mistake or it should be this way for some reasons, but in 

> Modelica.Fluid.Examples.HeatExchanger.BaseClasses.WallConstProps

both equations include multiplication by **2**. 

`heatPort_a[i].Q_flow=2*k_wall/s*(Ta[i]-T[i])*area_h/n;`
`heatPort_b[i].Q_flow=2*k_wall/s*(Tb[i]-T[i])*area_h/n;`

If it is not a mistake, could anyone explain why it is there?

beutlich, 2019-08-15:
See #1633. Does this help to answer your question?

Ciobrelis, 2019-08-15:
Yes, thank you.

Ciobrelis, 2019-09-04:
Sorry, to bother you again with this thread. But I still think that there is a problem with the multiplication by 2. 
In the entire code, the wall thickness s is assumed as a whole thickness, not as half.  
Therefore, multiplication by 2 essentially makes that the wall thickness is twice the real one instead of half. 

`heatPort_a[i].Q_flow=2*k_wall/s*(Ta[i]-T[i])*area_h/n;`
 ` heatPort_b[i].Q_flow=2*k_wall/s*(Tb[i]-T[i])*area_h/n;`

It should be rather multiplied by 0.5 instead of 2.

closed 2019-09-04 by Ciobrelis