modelica/ModelicaStandardLibrary #3099: Non-identical declarations in Media.Air.MoistAir

maltelenz, 2019-08-23:
In the class `ModelicaTest.Media.TestAllProperties.MoistAir`, `ModelicaTest.Media.TestAllProperties.PartialMediumFunctions` is extended, redeclaring the Medium `Modelica.Media.Interfaces.PartialMedium` to `Modelica.Media.Air.MoistAir`.

`Modelica.Media.Air.MoistAir` has the function `specificInternalEnergy`, with the declaration:

`output SI.SpecificInternalEnergy u "Specific internal energy"`

which resolves to:

`Real(final quantity = "SpecificEnergy", final unit = "J/kg")`

`Modelica.Media.Interfaces.PartialMedium` has this declaration in the corresponding `specificInternalEnergy` function:

`output SpecificEnergy u "Specific internal energy"`

which resolves to `Modelica.Media.Interfaces.Types.SpecificEnergy`:

`type SpecificEnergy = SI.SpecificEnergy(min = -1.0e8, max = 1.e8, nominal = 1.e6) "Type for specific energy with medium specific attributes";`

which in turn resolves to:

`Real(final quantity = "SpecificEnergy", final unit = "J/kg", min = -1.0e8, max = 1.e8, nominal = 1.e6)`

Do these declarations follow the requirement in 7.1, that declaration elements are "exactly identical to any element of the flattened enclosing class with the same name and the same level of protection (public or protected) and same contents".

1. Are they allowed to go separate ways through the lookup, or do they have to be syntactically identical?
2. The two paths don't have the same attributes. Is that allowed?

beutlich, 2019-10-06:
@hubertus65 Why are there two different types Modelica.Media.Interfaces.Types.SpecificEnergy and Modelica.SIunits.SpecificInternalEnergy? Why are they used inconsistently?

thorade, 2019-11-06:
As both variables are named `u` I would assume they really should both use SI.SpecificInternalEnergy.
https://en.wikipedia.org/wiki/Specific_internal_energy
Using a different type seems to have no benefit, and it is OK to change it to be consistent. I assume it was just not noticed because Dymola did not complain.

So, the real question here is not about thermodynamics, but about the language specification.

HansOlsson, 2019-11-06:
> As both variables are named u I would assume they really should both use SI.SpecificInternalEnergy.

I thought both should use the Media-package specific SpecificInternalEnergy (and not SI.) - since I thought the intent was to use them to allow different scaling.

> Using a different type seems to have no benefit, and it is OK to change it to be consistent. I assume it was just not noticed because Dymola did not complain.
> So, the real question here is not about thermodynamics, but about the language specification.

I agree that they don't seem identical and we should detect that. I believe the issue may be related to the redeclare extends.

maltelenz, 2019-11-06:
So we agree the model as it stands is illegal?

HansOlsson, 2019-11-06:
> So we agree the model as it stands is illegal?

Yes, as far as I can see.

PR #3204 (2019-11-07): Fix definition of specificInternalEnergy in MoistAir
Closes #3099.
PR #3209 (2019-11-08): Fix definition of specificInternalEnergy in MoistAir
Back-port of #3204 to fix #3099 in maint/3.2.3.
closed 2019-11-07 by beutlich