modelica/ModelicaStandardLibrary #3102: Fluid Examples. HT wall model.

Ciobrelis, 2019-08-28:
I have tried to reuse the wall HT model that is provided in the fluid examples. In the scenario that I have 5 volumes and I enable temperatures values at the start, I get an error. 

[image]

### **Error:**
[image]

I think there is missing some logic / code behind the model, that it would automatically use the fill(value,n) function. So in case, I add it, I get:

[image]
[image]

Which basically requires one to manually also add fill() function for Boolean. In this case to add fill(true,nNodes) in multiple lines. for T, Ta and Tb. 

[image]

Lastly, if you enter the wall model via Diagram view, all fill() functions for Boolean reset, and then one has to enter it again via Text view. 

I hope it is clear.

beutlich, 2019-08-28:
This is not specific to your application. Consider
```mo
model M
  parameter Real p[5];
end M;
```
You cannot pass a scalar to p, but need to provide a vector in all cases, for example using fill. So it either is a specification issue (if we allow assignments between arrays and scalars) or a tool issue (to better guide you in modeling comfort).

casella, 2019-09-07:
This is not a problem with the Modelica Standard Library, but rather a missing feature of OpenModelica, which has no way to add the `each` prefix to specify parameter arrays by just giving one value which is used for all elements of the array.

Please follow [#5405](https://trac.openmodelica.org/OpenModelica/ticket/5405) on the OpenModelica trac.

closed 2019-09-07 by casella