modelica/ModelicaStandardLibrary #3103: Annotations 'tab' and 'group' in Camel Case

tobolar, 2019-09-02:
Text indicated by the annotations **tab** is in some components Camel Case (e.g. _Internal Interface_ for `Medium` of `Modelica.Fluid.Machines.BaseClasses.PumpMonitoring.PumpMonitoringBase`), in some not (e.g. _Nominal resistances and inductances_ for `Lrsigma` in `Modelica.Electrical.Machines.BasicMachines.InductionMachines.IM_SlipRing`).

Similar mixing holds probably for the annotation **group** as well (I didn't check.).

beutlich, 2019-09-04:
Any preferences? I'd prefer to change "Internal Interface" to "Internal interface".

DagBruck, 2019-09-04:
Not the biggest problem in the world, but my #1 preference is consistency.

I think most design guidelines I have seen suggest writing it as a normal sentence, i.e. "Internal interface".

tobolar, 2019-09-05:
> ... my No.1 preference is consistency.

That's the point of this issue.

I have _slight_ preference to use Came Case in tabs and "sentence-like" style in groups.
Tab's text is mostly short, containing just a guide word(s). I've check in web some screenshots of other software and my impression is that this is handled quite differently:
https://docs.microsoft.com/en-us/windows/win32/uxguide/ctrl-tabs
https://www.trueos.org/blog/lumina-screenshot-simple-utility-powerful-results/
https://blog.cloudhq.net/send-screenshots/
https://www.tightvnc.com/screenshots.php

DagBruck, 2019-09-05:
> https://docs.microsoft.com/en-us/windows/win32/uxguide/ctrl-tabs

But that reference actually recommends "Use sentence-style capitalization", see the section Labels near the end. And their example follow that.

tobolar, 2019-09-05:
Fine to me. Their guidelines are quite clear and comprehensive so we could simply adopt it.

PR #3110 (2019-09-07): Consistent spelling of tab and group in Dialog annotation
close #3103
closed 2019-09-07 by beutlich