modelica/ModelicaStandardLibrary #3104: Icon of Modelica.Mechanics.MultiBody.Joints

tobolar, 2019-09-03:
Based on comments in #3087 on icon for `Joints` package, here is a concluded suggestion: model `IconForJoints.Joints4a` in https://gist.github.com/tobolar/95e62615d30ed5931cfd94c245040756

In Dymola: 
[image]

@maltelenz @henrikt-ma @beutlich: Please check if acceptable rendering can be achieved in other tools.

beutlich, 2019-09-03:
Here's the screen shot from SimulationX

[image]

But there is a typo rotatio -> rotation in line 9. And there is invalid inheritance in Joints4a and Joints7.

tobolar, 2019-09-03:
> But there is a typo rotatio -> rotation in line 9. And there is invalid inheritance in Joints4a and Joints7.

I will fix it eventually for PR.

maltelenz, 2019-09-03:
SystemModeler:
[image]
OpenModelica:
[image]

maltelenz, 2019-09-03:
And for good measure, WL/Mathematica:

[image]

tobolar, 2019-09-03:
OK, I think the tools render well the variants with contours. I will prepare PR with Joints4a icon.

PR #3106 (2019-09-04): Better icon for MultiBody.Joints
close #3104
closed 2019-09-04 by beutlich