modelica/ModelicaStandardLibrary #3115: Derivative of step signal by CombiTimeTable

beutlich, 2019-09-10:
Discontinous signals can/should **not** be differentiated:

```mo
model M1
  Real y=if time <= 0.5 then 0 else 1;
  Real yd=der(y);
end M1;
```

This has been considered in #1147 when implementing the constant interpolation of CombiTimeTable:

```mo
model M2
  Real yd=der(combiTimeTable.y[1]);
  Modelica.Blocks.Sources.CombiTimeTable combiTimeTable(
    table=[0,0;0.5,1;1,1],
    smoothness=Modelica.Blocks.Types.Smoothness.ConstantSegments);
end M2;
```

However, using linear interpolation of CombiTimeTable and duplicated sample points we successfully can model a discontinuous signal and get its derivative.

```mo
model M3
  Real yd=der(combiTimeTable.y[1]);
  Modelica.Blocks.Sources.CombiTimeTable combiTimeTable(
    table=[0,0;0.5,0;0.5,1;1,1]);
end M3;
```

This seems inconsistent, if not even wrong. Since linear interpolation with duplicated sample points was one of the designed features of the CombiTimeTable it would be a major loss if we'd deprecate it. I have no good idea how to solve this inconsistency and would like to get @HansOlsson's and @MartinOtter's feedback,

MartinOtter, 2019-09-11:
In Modelica it is possible to model a discontinuous signal and then differentiate it as shown in your example M1. The differentiation of a table is in a similar spirit. In general, it seems hard (impossible) to detect such issues at compile time and therefore this needs to be detected during simulation. This in turn means that this is a tool issue (if a differentiation would lead to a Dirac impulse, and this is not supported by the tool, then at least a warning message needs to be printed). As I recall, in fact Dymola can detect some cases of this kind during run-time.

beutlich, 2019-09-11:
Yes, Dymola simulation fails for models M1 and M2 at time 0.5s. With the knowledge of #1147 this is as expected. However, M3 models the same step signal using the same block (CombiTimeTable) but shows different behaviour when differentiating. I'd like to fix it (or at least document it) if @HansOlsson and @MartinOtter also think that is misuse of the linear interpolation with duplicated sample points.

PR #3370 (2020-01-27): Document the availability of derivatives of table blocks
Document the availability of first and second derivatives, its exceptions and the design inconsistency reported by #3115.

Resolves #3115.
closed 2020-01-27 by beutlich