modelica/ModelicaStandardLibrary #3121: Null Pointer Dereference on Error in ModelicaIInternal_readDirectory

joehenning, 2019-09-18:
I found this with a static analysis tool while troubleshooting an unrelated bug. I think that practically speaking, it would be really rare to come across this problem.

```
            /* Allocate Modelica memory for file/directory name and copy name */
            pName = ModelicaAllocateStringWithErrorReturn(strlen(pinfo->d_name));
            if ( pName == NULL ) {
                errnoTemp = errno;
                closedir(pdir);
                if ( errnoTemp == 0 ) {
                    ModelicaFormatError("Not possible to get file names of \"%s\":\n"
                        "Not enough storage", directory);
                }
                else {
                    ModelicaFormatError("Not possible to get file names of \"%s\":\n%s",
                        directory, strerror(errnoTemp));
                }
            }
            strcpy(pName, pinfo->d_name);
```

If ModelicaAllocateStringWithErrorReturn were to return an error, pName would be NULL and the code would go down the error handling code branch, which logs an error. After the error handling branch, the code tries to strcpy into pName, even if it is NULL. Correct code should cause the function to return at the end of the `if(pName == NULL)` block, rather than trying to write into pName when it is NULL.

HansOlsson, 2019-09-18:
>  After the error handling branch, the code tries to strcpy into pName, even if it is NULL. 

But this is not actually an issue, since ModelicaFormatError triggers error handling, and thus the code should not continue to the following lines in that case. 

I don't know if there are some "annotations" missing on ModelicaFormatError for the C-code analysis.

joehenning, 2019-09-18:
Ahh, both myself and the tool overlooked the `exit(1)` in the definition of ModelicaFormatError. Thanks for the prompt reply. I will close the issue.

If I can trouble you for a further question though, does this mean exported models will sometimes close the process that's running them? For example, will an application running an FMU unexpexctedly close whenever ModelicaFormatError is called? That's certainly a sensible response to the out-of-memory condition being handled above, as there's little chance for a graceful shutdown, but for other error cases this is something I may need to pay more attention to in my own work.

closed 2019-09-18 by joehenning