modelica/ModelicaStandardLibrary #3122: Possibly switch Modelica documentation from HTML to Markdown

christiankral, 2019-09-21:
The creating of HTML documentation in any Modelica library is very fault prone (e.g. https://github.com/modelica/ModelicaStandardLibrary/pull/3054/commits/5ae807340a533535d6733b39edffca1b63883817) as most documentation is created manually. Therefore, we continuously run into wrong or missing tags. Writing documentation in Markdown format is lot easier, faster and less fault prone.

So in the long run it may make sense to switch from HTML to Markdown in the Modelica documentation as suggested in https://github.com/modelica/ModelicaStandardLibrary/pull/2867#issuecomment-480893406. This is an idea I wanted to share as some others may already have thought about it.

sjoelund, 2019-09-21:
It would be very easy to swap the libraries to markdown since markdown allows you to mix html and markdown (as long as tags are closed, etc)

dietmarw, 2019-09-21:
In don't think this is a question for the library but first of all of tool support. Should tools support be in place then I'm all for it. We had something similar regarding MathJax support. Which is no problem in OM for example but not in other tools.

christiankral, 2019-09-21:
One way to go could be to have something like a Modelica Markdown editor and viewer (plugin or whatever) which can be used from all tool vendors. This Modelica Markdown editor/viewer should then be open source and the initial version may even be paid work financed by the Modelica Association. From such an initial version of this editor/viewer the tool vendors could improve and maintain the editor/viewer source code. 

The advantage of a common code basis were that the particular (Modelica specific) Markdown flavor does not have to be re-coded by every tool vendor.

HansOlsson, 2019-09-23:
One could argue that we planned ahead for this by having `<html></html>` surrounding the current documentation.

> One way to go could be to have something like a Modelica Markdown editor and viewer (plugin or whatever) which can be used by all tool vendors. 

I don't see that as a good idea for several reasons: different tools have different plugin-capabilities, and tools also want the documentation editor to have the same look-and-feel as the rest of the GUI that will require substantial configuration possibilities in the plugin, which makes it a lot more complicated project to manage.

And if we go for markdown we need to decide on which one, and I would prefer that we go for a standard one instead of inventing an Modelica-specific one. (Since it seems to be less work than reinventing the wheel, and it would be good to break out of mindset of MA-specific solutions.)

And in that case I don't see a need for developing a new plugin within MA, since it seems likely that there already exists plugin(s) for that.

henrikt-ma, 2019-09-23:
This feels related to our ongoing work with the prototype for predefined plots.  As we will show at the upcoming design meeting, our current prototype doesn't make use of HTML for several reasons, which I'm sure lead to discussions giving us the chance to explain in more detail next week in Lund.

beutlich, 2019-09-23:
> The creating of HTML documentation in any Modelica library is very fault prone (e.g. [5ae8073](https://github.com/modelica/ModelicaStandardLibrary/commit/5ae807340a533535d6733b39edffca1b63883817)) as most documentation is created manually. Therefore, we continuously run into wrong or missing tags.

It would be incredibly helpful if we could validate the HTML in a CI job.

beutlich, 2022-03-11:
> It would be incredibly helpful if we could validate the HTML in a CI job.

This is the case since quite some time.

dietmarw, 2022-06-15:
Probably interesting to note that Dymola 2023 now supports Markdown and LaTeX (via MathJax) out of the box. Only thing that one has to do is to switch the rendering mode to Markdown. What is also interesting is that the LaTeX rendering also works in HTML mode (though not documented, @HansOlsson ? )