modelica/ModelicaStandardLibrary #3124: Possible discrepancy between parameter values in reference book vs function implementation in Modelica.Fluid.Dissipation.PressureLoss.Bend.dp_edgedOverall_DP

tamer3DS, 2019-09-27:
In reference to function **dp_edgedOverall_DP**
SOURCE_1: Idelchik, I.E.: HANDBOOK OF HYDRAULIC RESISTANCE, 3rd edition, 2006

In line 67: Real B=24.8
    "Coefficient considering effect of Reynolds number on zeta_TOT";

And In line 69:   Real exp=0.263
    "Exponent for Reynolds number correction in laminar regime";

But according the textbook (image attached), the coefficients were expected to be:
B=4.06
exp=0.118

Can you please clarify the source of the values used in the function?

[image]

bilderbuchi, 2019-09-30:
Without having the source available (and thus the geometry in question), one thing that jumps out to me is that the data in the screenshot you give is valid for Re>=1e4 (which sound like in the turbulent regime to me), but the comment in the source speaks about "laminar regime".

tamer3DS, 2019-09-30:
I am sorry I didn't make my question clear.
**The image attached is from page 365 from "source 1: Idelchik, I.E.: HANDBOOK OF HYDRAULIC RESISTANCE, 3rd edition, 2006"** --> I could not find the source for the values used in the function starting line 66
`//SOURCE_1: p.365: Correction w.r.t effect of Reynolds number in laminar regime`
`Real B=24.8`
    `"Coefficient considering effect of Reynolds number on zeta_TOT";`
`Real exp=0.263`
    `"Exponent for Reynolds number correction in laminar regime";`

Can you please help me explain where the values used in these two coefficients (B and exp) come from?
Thanks for your help with this

beutlich, 2019-12-04:
@casella Can you help out here? Do you have the cited 3rd edition of Idelchick available?

casella, 2019-12-04:
I checked my library, we have the 2nd edition, 1986, the 3dr edition, 1996, and the 4th edition, 2007. I'm not sure which one @wischhusen referred to when he wrote the Fluid_Dissipation library. AFAIK there is no 3rd edition 2006.

Some input from @wischhusen would be helpful, I guess they used the same source for the whole library but I don't know what that was.

wischhusen, 2019-12-05:
Hi all,
have to check. The factors are used here. It could be, that the reference for the factors is Source_2 instead.

  //SOURCE_2: p.208, diag. 9.3: Correction w.r.t effect of Reynolds number
  Real CF_Re=SMOOTH(
      Re_turb_min,
      Re_turb_max,
      Re)*B/Re^exp + SMOOTH(
      Re_turb_max,
      Re_turb_min,
      Re) "Correction factor for Reynolds number";

casella, 2019-12-05:
Thank you @wischhusen! I would gladly help, but getting up to speed with the inner details of the library would take a lot of time to me - I guess it's much easier to you that know the thing from the inside out.

wischhusen, 2019-12-16:
[image]

Unfortunately, the page number was wrong. It is p. 142 in Miller's "Internal Flow Systems". Fig. 9.3 is correct. The factor B and exp is derived to fit the value of C_Re at Re=1e4 (see image). Probably these factors were chosen since the range is not limited to Re>=1e4 as in Idelchik.

beutlich, 2019-12-16:
> AFAIK there is no 3rd edition 2006.

> Unfortunately, the page number was wrong. 

@wischhusen Can you please create a pull request to correct both sources. Thanks.

beutlich, 2020-05-10:
@wischhusen Any update on the pull request?