modelica/ModelicaStandardLibrary #3127: Wrong unit in ModelicaTest.Fluid.Dissipation.Verifications.PressureLoss.Bend examples

carlj-w, 2019-10-01:
In ModelicaTest.Fluid.Dissipation.Verifications.PressureLoss.Bend.dp_curvedOverall_DPMFLOW we have this line:
`parameter Modelica.SIunits.Conversions.NonSIunits.Angle_deg delta[3] = {30, 45, 90} * Modelica.Constants.pi / 180 "Angle of turning";`
and in ModelicaTest.Fluid.Dissipation.Verifications.PressureLoss.Bend.dp_edgedOverall_DPMFLOW this line:
`parameter Modelica.SIunits.Conversions.NonSIunits.Angle_deg delta[4] = {30, 45, 90, 180} * Modelica.Constants.pi / 180 "Angle of turning";`

it seems to me that both right hand sides converts angles from degrees to radians, so the result is in radians. However, the variables on the left hand sides have their units set to degrees.

wischhusen, 2019-10-02:
Correct. Should be SIunits.Angle.

dietmarw, 2019-10-02:
@wischhusen If you do not mind then I like to use the opportunity of this trivial bug to create a screencast of how to fix issues in the MSL and how to submit a PR. So simply leave this to me :-)

PR #3301 (2020-01-17): Fixes #3127 by giving the parameter the correct type
Sorry I won't have time to make a tutorial out of this one.

Fixes #3127.
commit 629621bc (2020-01-17): Fixes #3127 by giving the parameter the correct type (#3301)
closed 2020-01-17 by dietmarw