modelica/ModelicaStandardLibrary #3132: Inconsistencies between engines of Mechanics.MultiBody.Examples.Loops

christoff-buerger, 2019-10-04:
At the Modelica conference this year I presented a language extension for non-monotonic modeling (cf. [Modelica language extensions for practical non-monotonic modelling: on the need for selective model extension](https://modelica.org/events/modelica2019/proceedings/html/papers/Modelica2019paper3B1.pdf)). One of its running examples are the `Engine1a`, `Engine1b` and `Engine1b_analytic` models of `Mechanics.MultiBody.Examples.Loops` I used to demonstrate how easy consistency problems are introduced. The idea of these models is to design a final analytic engine model starting from an idealized model via one considering the gas force in the cylinder. They are inconsistent however, most likely because of _historically_ individual model-changes that have not been properly propagated throughout the whole design scenario.

The objective of this ticket is to fix these inconsistencies **by the ordinary means of Modelica 3.4 (i.e., not selective model extension, but simply manual fixing)**.

christoff-buerger, 2019-10-04:
I will provide a consistent modeling from my MSL fork via a pull request.

christoff-buerger, 2019-10-04:
Added respective pull request #3132.

dietmarw, 2019-10-04:
Just for the future. It is not necessary to open up an issue in addition of a PR. The PR can be discussed in the PR itself while keeping the number duplicate issue numbers down.

PR #3133 (2019-10-09): Made engine designs consistent w.r.t. the whole design scenario
This pull request fixes the inconsistencies between the `Engine1a`, `Engine1b` and `Engine1b_analytic` models of `Mechanics.MultiBody.Examples.Loops`. For details about these, please cf. Section 4.2 of [Modelica language extensions for practical non-monotonic modelling: on the need for selective model extension](https://modelica.org/events/modelica2019/proceedings/html/papers/Modelica2019paper3B1.pdf).

The pull request fixes #3132.

The changes are:
1. Consistent stop times:
   - Engine1b, Engine1b_analytic: stop time set from 0. [... cut]
closed 2019-10-09 by beutlich