modelica/ModelicaStandardLibrary #3143: .Modelica.Utilities.System.getTime() returns bad year

adrpo, 2019-10-10:
Running this model: 
```
model GetTime
  Integer ms;
  Integer sec;
  Integer min;
  Integer hour;
  Integer mon;
  Integer year;
algorithm
  (ms, sec, min, hour, mon, year) := .Modelica.Utilities.System.getTime();
  .Modelica.Utilities.Streams.print("ms:" + String(ms) + "\n");
  .Modelica.Utilities.Streams.print("sec:" + String(sec) + "\n");
  .Modelica.Utilities.Streams.print("min:" + String(min) + "\n");
  .Modelica.Utilities.Streams.print("hour:" + String(hour) + "\n");
  .Modelica.Utilities.Streams.print("mon:" + String(mon) + "\n");
  .Modelica.Utilities.Streams.print("year:" + String(year) + "\n");
end GetTime;
```
with OpenModelica on Windows will give you `year=10` and on Linux will give you `year=9`.
With Dymola 2019 on Windows ill give you `year=10`.

The code for the external C is here:
https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Resources/C-Sources/ModelicaInternal.c#L1264
I haven't looked in detail into it, maybe somebody can spot the problem.

adrpo, 2019-10-10:
Reference: https://stackoverflow.com/questions/58320730/modelica-print-current-time/

beutlich, 2019-10-10:
Hm, only for Win32 we increment the month

https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Resources/C-Sources/ModelicaInternal.c#L1306

sjoelund, 2019-10-11:
I guess the problem with easily assigning to the bad value is due to not returning enumeration (or wrapping it into a record).

HansOlsson, 2019-10-11:
> I guess the problem with easily assigning to the bad value is due to not returning enumeration (or wrapping it into a record).

Wrapping it into a record seems the most logical.

Alternatively I think that the risk would have been less if we had used the same order as IS 8601 and reversed the output order to return year, month, ... (but changing it now would be too confusing).

adrpo, 2019-10-11:
Could we change it to a record for 4.x?

sjoelund, 2019-10-11:
I think there might be a possibility to rename getTime() to Internal.getTime() or something with the same interface. And then create a new getTime() that works by calling that function and creating a record on top of it.

thorade, 2019-10-11:
+1 on ISO 8601 for dates everywhere.
https://en.wikipedia.org/wiki/ISO_8601
Would such a breaking change be allowed for MSL 4.0?

closed 2019-10-11 by beutlich

beutlich, 2019-12-02:
> 
> 
> I think there might be a possibility to rename getTime() to Internal.getTime() or something with the same interface. And then create a new getTime() that works by calling that function and creating a record on top of it.

This is addressed by #3247.

PR #3144 (2019-10-11): Fix month and year correction in ModelicaInternal_getTime
Closes #3143.
PR #3147 (2019-10-15): maint/3.2.3: Fix month and year correction in ModelicaInternal_getTime
back-port of #3144 to fix #3143 in maint/3.2.3.