modelica/ModelicaStandardLibrary #3150: Invalid modifiers in Modelica.Magnetic.QuasiStatic.FundamentalWave examples.

perost, 2019-10-15:
The following modifications of `m` are invalid, because `m` in [`DQToThreePhase`](https://github.com/modelica/ModelicaStandardLibrary/blob/4a45701108d2343ea0a90c5b68c0e0f53f790c07/Modelica/Electrical/Machines.mo#L17218) is declared as a constant, whereas the `m`:s declared in these two models are declared as parameters:
https://github.com/modelica/ModelicaStandardLibrary/blob/4a45701108d2343ea0a90c5b68c0e0f53f790c07/Modelica/Magnetic/QuasiStatic/FundamentalWave/Examples/BasicMachines/SynchronousMachines/SMPM_CurrentSource.mo#L65-L66

https://github.com/modelica/ModelicaStandardLibrary/blob/4a45701108d2343ea0a90c5b68c0e0f53f790c07/Modelica/Magnetic/QuasiStatic/FundamentalWave/Examples/BasicMachines/SynchronousMachines/SMR_CurrentSource.mo#L33-L34

beutlich, 2019-10-15:
This bug was introduced as side-effect of 891d6c34087cc20954c6c8a6b331341e5aece90f for #3058. Instead of renaming also the variability of `m` was changed.

henrikt-ma, 2019-10-17:
In my opinion, the correct way to solve this is not to change constants to parameters, but rather the opposite; all the `m` involved here should be declared `constant`.  This makes it clear that they are not intended for change after translation, and variability rules are respected.

henrikt-ma, 2019-10-23:
As this was closed via #3152, can you please confirm that it was intentional to resolve this in a way that works poorly with tools that don't eagerly evaluate parameters?

**Quick background for anyone who hasn't been following related issues and pull requests:** The treatment of Modelica parameters in SystemModeler is basically different from the treatment of Modelica constants in the same way as parameters and constants are really two different things in FMI; a parameter can be changed after translation (construction of the FMU), a constant can't.  This difference makes perfect sense and is very valuable for understanding the intent and possible configuration modes of a model/FMU.  Of course, we have to compromise this basic rule due to "structural parameters" not being declared constant, and due to `Evaluate=true`, but at least we try to go in the direction of aligning the concepts in Modelica and FMI by respecting the difference between parameters and constants.

By now, I am well aware that not all Modelica tools share this ambition, and wouldn't be surprised to see developments (such as how this issue was resolved) going in the other direction.

In the particular case of magnetic machines, the need to avoid parameters is explained in more detail in #1966 and pull/3119.

Reopening while waiting for the intent of the resolution be confirmed.  My apologies if the situation is already well understood by everybody involved in pull/3152.

beutlich, 2019-10-23:
> can you please confirm that it was intentional to resolve this

Yes, it was closed intentionally since it was a side-effect regression by 891d6c34087cc20954c6c8a6b331341e5aece90f when implementing #3058. And it resolves the @perost's primary request to fix the invalid modifier.

What you prefer is clearly an enhancement as @AHaumer tried to do in his commit inconsistently. I am all in favour to get this resolved in a clean way. Please open a new issue for it and assign it to @AHaumer and @christiankral.

PR #3152 (2019-10-21): Fix variability from constant to parameter
Closes #3150.
closed 2019-10-23 by beutlich