modelica/ModelicaStandardLibrary #3162: Unusual Constant Source Voltage icon in 3.2.3

MartinOtter, 2019-10-20:
In MSL up to version 3.2.2, the icon for a constant source voltage was at it is standard in the electrical community:
[image]

In MSL 3.2.3 the icon was changed to the non-standard form:
[image]

Can someone explain, why this change was made.

beutlich, 2019-10-20:
See #2718.

beutlich, 2019-10-20:
See #2186.

christiankral, 2019-10-22:
Some simulators, e.g. LTSpice,

[image]

only use the circular symbol, so on my opinion the change is OK.

beutlich, 2019-12-04:
Closing after silent acceptance.

closed 2019-12-04 by beutlich