modelica/ModelicaStandardLibrary #3164: Add an arrow to SweptVolume?

sergniko, 2019-10-22:
Is it a good idea to add an arrow to **Fluid.SweptVolume** as for most of **Translational.Components**?
In class documentation it is said: "The force of the flange results from the pressure difference between medium and ambient pressure and the cross sectional piston area". But nothing about sign convention.
For me, as a new user, it took a lot of time to understand why the force is negative while the pressure is above ambient pressure.
Adding arrow can help to get better understanding. For example **Translational.Examples.WhyArrows** is very helpful.

beutlich, 2019-10-22:
@christiankral As we no longer have active library officers for Fluid and as @sergniko referred to Mechanical.Translational, could you please take care of it? Thanks a lot.

christiankral, 2020-02-15:
I somehow missed this issue because of the focus on MSL 4.0.0.

From investigating `ModelicaTest.Fluid.TestComponents.Machines.TestSweptVolume` 

[image]

[image]
 
So I had a look at `Modelica.Fluid.Machines.SweptVolume` and created PR #3428 to include an arrow.

dietmarw, 2020-02-17:
fixed in #3428

closed 2020-02-17 by dietmarw

sergniko, 2021-02-07:
The arrow is missing again in latest build/release.