modelica/ModelicaStandardLibrary #3165: Wrrong position of fire input in GTO

max-privato, 2019-10-22:
The GTO fire input should be placed in contact or very near the actual gate wire. The present switch, instead, has the fuchsia fire input too far from the gate blue indicator: see the image below.

[image]

dietmarw, 2019-10-22:
Just as a comment this position was probably chosen by design. Also present in `IdealThyristor`. 
Other components that have a firing signal (e.g., `SimpleTriac`) also route it to the corner. Though they draw also a connection line on the icon. So I would be careful calling this "wrong" but rather discuss if it should be improved or not.

max-privato, 2019-10-22:
I totally agree with @dietmarw position.

I opened this ticket not because I thought something was wrong with the symbol.
Instead, I just evaluated the actual effectiveness of the diagrams.
Consider for instance the following diagram: I think that a connection of the fuchsia line to the gate would surely significantly enhance the diagram appearance and quality.
If we don't want to modify the position of the fuchsia arrow, we could just add a connection line.

[image]

dietmarw, 2019-10-23:
What I could think of is adding a dashed line so these would look like:
[image]
Without changing  the position of the connector.
What do you think?

dietmarw, 2019-10-23:
One thing I *did* notice is that the symbol of the thyristor is actually that of a GTO, i.e. the bar across should not be there. Here is what I would change the symbols to so that they actually are more correct and clean:
[image]
Again, what do you think?

max-privato, 2019-10-23:
I agree with both your proposals: the connecting lines for thyristor and GTO form me are ok.
I also agree that the thyristor symbol with the bar across is that of a GTO and therefore should be corrected.

dietmarw, 2019-10-23:
PR created that changes the design to the following:
[image]

PR #3169 (2019-10-29): Fix #3165 by improving the icon design of the firing connector
fix #3165
commit 7a963bcc (2019-10-29): Merge pull request #3169 from dietmarw/t3165-trigger

Fix #3165 by improving the icon design of the firing connector
closed 2019-10-29 by beutlich