modelica/ModelicaStandardLibrary #3168: Missing concept for handlig the initial conditions of capacitances

beutlich, 2019-10-23:
See https://github.com/modelica/ModelicaStandardLibrary/issues/2146#issuecomment-387703451.

christiankral, 2019-10-29:
The title of this ticket suggests that we have a general problem with initialization of capacitors. I suppose this is not the case. How I understand the ticket, this issue is related with the NPN transistor models in the package `Analog.Semiconductors`. 

@beutlich Is this correct? If yes we may consider to update the title of this ticket.

beutlich, 2019-10-30:
> How I understand the ticket, this issue is related with the NPN transistor

https://github.com/modelica/ModelicaStandardLibrary/issues/2146#issuecomment-387703451 from @kristinmajetta reads:

> This concerns other models containing capacitances besides the NPN model.

To me it sounded that it is a rather general issue.

beutlich, 2019-11-13:
I believe, the unused parameter `UIC` of Modelica.Electrical.Spice3.Basic.C_Capacitor also belongs here.