modelica/ModelicaStandardLibrary #3172: Undefined name in annotation in Mechanics.MultiBody.Visualizers.Torus

eshmoylova, 2019-10-24:
In Modelica.Mechanics.MultiBody.Visualizers.Torus in the following annotation `multiColoredSurface` is used. The name is not defined in the model or in the class from which it extends.
https://github.com/modelica/ModelicaStandardLibrary/blob/6e6180f71adeaeb6257081f226034dff92963848/Modelica/Mechanics/MultiBody/Visualizers.mo#L868-L869

eshmoylova, 2019-10-24:
The example model Modelica.Mechanics.MultiBody.Examples.Elementary.Surfaces fails in Dymola pedantic mode due to this unresolved name. Note that the example also uses VoluminousWheel component that has issue #3173 and should fail in the pedantic mode until that issue is fixed.

HansOlsson, 2019-10-24:
This is similar to #2004 but for a different visualizer.

PR #3174 (2019-10-24): Remove undeclared variable from Dialog.enable
closes #3172
PR #3175 (2019-10-25): Remove undeclared variable from Dialog.enable
Back-port #3174 to fix #3172 in maint/3.2.3.
closed 2019-10-24 by beutlich