modelica/ModelicaStandardLibrary #3173: Negative radius in Mechanics.MultiBody.Visualizers.VoluminousWheel

eshmoylova, 2019-10-24:
The default value for `ri` parameter in Modelica.Mechanics.MultiBody.Visualizers.VoluminousWheel is negative:
https://github.com/modelica/ModelicaStandardLibrary/blob/6e6180f71adeaeb6257081f226034dff92963848/Modelica/Mechanics/MultiBody/Visualizers.mo#L969

Given the default values of other parameters, the value for `ri` is -0.05. Modelica.SIunits.Radius has `min = 0`. `ri` is supposed to be used as an inner radius for a torus component.

The example model Modelica.Mechanics.MultiBody.Examples.Elementary.Surfaces uses the wheel with the default parameters, so `ri` there is negative.

beutlich, 2019-10-24:
@dzimmer @MartinOtter How is it supposed to work?

tobolar, 2019-10-25:
Negative values for `ri` shall be enabled. It is just a virtual radius. Proposal: use `SI.Length` instead of `Radius`.

beutlich, 2019-10-25:
OK. Can do.

tobolar, 2019-10-25:
@beutlich PR is yet ready. :-)

PR #3176 (2019-12-04): Change limits of parameter ri
Close #3173
PR #3250 (2019-12-05): Change limits of parameter ri of VoluminousWheel
Back-port of (necessary changes of) #3176 to fix #3173 in maint/3.2.3.
closed 2019-12-04 by beutlich