modelica/ModelicaStandardLibrary #3177: Confusing names for torus parameters

eshmoylova, 2019-10-28:
Modelica.Mechanics.MultiBody.Visualizers.Advanced.SurfaceCharacteristics.torus ans Modelica.Mechanics.MultiBody.Visualizers.Torus use confusing names for the description of the parameters: `ri` and `ro`.
Visualizers.Torus defines:
https://github.com/modelica/ModelicaStandardLibrary/blob/491ccf7885b8c23ba94dda5c9cba61b26b2a932d/Modelica/Mechanics/MultiBody/Visualizers.mo#L857-L858

And SurfaceCharacteristics.torus defines:
https://github.com/modelica/ModelicaStandardLibrary/blob/491ccf7885b8c23ba94dda5c9cba61b26b2a932d/Modelica/Mechanics/MultiBody/Visualizers.mo#L2450-L2451

Usually, the inner radius for a torus is usually defined as the radius of the hole and the outer one as the radius of the whole shape. For example, see [this](https://www.quora.com/How-would-you-find-the-volume-of-a-solid-torus-donut-shape). So having inner radius to be greater than the outer one does not make sense.

Looking at the formula used in SurfaceCharacteristics, it looks like `ri` is used as the major radius and `ro` as the minor radius, as defined [here](https://en.wikipedia.org/wiki/Torus].

It would be good if there was perhaps some reference given to how `ri` and `ro` are supposed to be interpreted.

MartinOtter, 2019-11-13:
Simplest probably to change the description texts.

- ri: Distance from the center of the tube to the center of the torus,
- ro: Radius of the tube.

Would be good to add a figure to make this clear.

I agree, the naming of the parameters could be better. However, names such as major/minor radius are also not completely obvious.

tobolar, 2019-11-22:
> - ri: Distance from the center of the tube to the center of the torus,
> - ro: Radius of the tube.

The description is fine. We can even use:
- ri: Major radius: distance from the center of the tube to the center of the torus,
- ro: Minor radius: radius of the tube.

Unfortunately, the parameter names `ri`and `ro`are confusing then since they imply thinking of inner and outer radii which is definitely false. Following [Wikipedia](https://en.wikipedia.org/wiki/Torus#Geometry), we could use 
- R: Major radius: distance from the center of the tube to the center of the torus,
- r: Minor radius: radius of the tube.

Unfortunately, [WolframMathWorld](http://mathworld.wolfram.com/Torus.html) uses exactly those names for indicating the inner and the outer radius. :-(
Thus
> Would be good to add a figure to make this clear.

is a must in this case.

beutlich, 2019-11-23:
I am in favour to follow Wikipedia and go for:

* R: Major radius (distance from the center of the tube to the center of the torus)
* r: Minor radius (radius of the tube)

beutlich, 2019-12-04:
I believe that this also was resolved by #3176.

closed 2019-12-04 by beutlich