modelica/ModelicaStandardLibrary #3184: Questionable display unit for Modelica.SIunits.Pressure

svorkoetter, 2019-10-30:
The display unit for `Modelica.SIunits.Pressure` is currently "bar". My colleague points out that:

> From https://en.wikipedia.org/wiki/Bar_(unit):

> The US National Institute of Standards and Technology (NIST) deprecates its use except for "limited use in meteorology" and lists it as one of several units that "must not be introduced in fields where they are not presently used".[4] The International Astronomical Union (IAU) also lists it under "Non-SI units and symbols whose continued use is deprecated".[5]

Should the display unit perhaps be changed to "Pa"?

bilderbuchi, 2019-10-31:
Hm, the same article also says
> The International Bureau of Weights and Measures (BIPM) lists the bar as one of the "non-SI units [that authors] should have the freedom to use", but has declined to include it among the "Non-SI units accepted for use with the SI".[1] 
The bar has been legally recognised in countries of the European Union since 2004.

so opinions seem to be split.

Also
> Many engineers worldwide use the bar as a unit of pressure because, in much of their work, using pascals would involve using very large numbers. In measurement of vacuum and in vacuum engineering, residual pressures are typically given in millibar, although torr or millimeter of mercury (mmHg) were historically common. 

and from my practice I can confirm both statements. In fact, the primary issue I experience is that one really has to be careful about clarifying which of `bar(g)` and `bar(a)` is used.

Another point is that, for ease of plotting, I try to make it so that all plotted variables are scaled as near as possible to ~ O(1), and I seldomly encounter pressures of O(1 Pa). Plotting quantities of, e.g., 1e5 Pa and 1 kg/s does not yield very useful default results.

I don't have a strong opinion on this. Maybe a useful compromise could be to make the displayUnit `MPa` (10 bar) by default?

HansOlsson, 2019-10-31:
> I don't have a strong opinion on this. Maybe a useful compromise could be to make the displayUnit MPa (10 bar) by default?

I don't think we should try to invent a new standard for pressure, but adapt to existing practice - and unfortunately that is a bit of a mess. And the problem with MPa is that normal atmospheric pressure is around 0.1 MPa which is slightly awkward.

And bonus fact for the mathematicians:
Bar pressure is also used within the world of coffee in almost all espresso machines, where nine bars of pressure is considered the canonical shot.[13]

bilderbuchi, 2019-10-31:
> 0.1 MPa which is slightly awkward

yeah, but 1e5 Pa seems even more awkward, no? 😄

HansOlsson, 2019-10-31:
> 0.1 MPa which is slightly awkward
> 
> yeah, but 1e5 Pa seems even more awkward, no? 😄

True, but there is also 100 kPa; but that isn't popular either.

svorkoetter, 2019-10-31:
My vote would be for kPa. That is how weather is reported here anyway. 100kPa isn't that far away from unity.

dietmarw, 2019-10-31:
Given that this ticket was originally about the concern "bar" no longer being allowed as SI unit and this actually not being the case I would be very careful changing the default display. In the engineering domain "bar" is still extremely common used unit. Changing this would have a negative impact in my opinion.

dietmarw, 2019-11-28:
I'm closing this as "wontfix" for now.

closed 2019-11-28 by dietmarw