modelica/ModelicaStandardLibrary #3201: Need to normalize input vectors to Modelica.Mechanics.MultiBody.Parts.FixedRotation if vector length small

hhammondscott, 2019-11-05:
One of our customers reported this issue to us. They were creating a inboard anti-roll bar model from components in the Modelica standard library. To get the mechanism to initialise in the correct arrangement they needed to normalize the vector inputs, **n_x** and **n_y**, to the Modelica.Mechanics.MultiBody.Parts.FixedRotation named **blade_frame_b** in the mechanism. The customer believe that this is because the lengths of the vectors n_x and n_y are small (10^-4 and 10^-6) and this causes a problem in the function  Modelica.Mechanics.MultiBody.Frames.from_nxy which is used in the FixedRotation model unless the input vectors are normalized.  There is nothing in the documentation to say this could be an issue. 
Attached is the example which used MSL version 3.2.3, and these were simulated with Dymola 2020. 
[KinematicLoopTest_Issue.zip](https://github.com/modelica/ModelicaStandardLibrary/files/3808605/KinematicLoopTest_Issue.zip)
The experiment **Modelica_test** shows the results of the model where the vectors **n_x** and **n_y** of the component **blade_frame_b** are not normalized. If you look at the animation you will see that the two sides of the mechanism start in different position. 
[image]

The experiment **Modelica_test_Normalize** shows the results when the vectors are normalized, in the animation both sides start in the same position. 
[image]

beutlich, 2019-11-05:
I can reproduce with MSL v3.2.3. As already reported, it all boils down to that the created orthogonal system in Modelica.Mechanics.MultiBody.Frames.TransformationMatrices.{from_nxy, from_nxz} (even if the input vectors are already orthogonal) does not correctly consider the vector length.

```mo
model T3201 "PoC for normalize issue with from_nxy"
  import Modelica.Math.Vectors.normalize;
  import Modelica.Mechanics.MultiBody.Frames.TransformationMatrices.Orientation;
  import Modelica.Mechanics.MultiBody.Frames.TransformationMatrices.from_nxy;
  parameter Real r_p17_b[:] = {0.1155,0,0.36417};
  parameter Real r_p19_b[:] = {0.1155,0.035,0.36417};
  parameter Real r_p20_b[:] = {0.046045,0.035,0.393641};
  // Create orientation from orthogonal vectors n_x and n_y
  parameter Real n_x[:] = cross(r_p17_b - r_p19_b, r_p20_b - r_p19_b) "={-0.001031485,0,-0.002430925}";
  parameter Real n_y[:] = cross(r_p20_b - r_p19_b, n_x) "={0,-0.00019923879031,0}";
  Orientation T_n = from_nxy(n_x, n_y) "={{-0.390608730963647,0,-0.920556798516511},{0,1,0},{0.920556798516511,0,-0.390608730963647}}";
  // Create orientation from orthonormal vectors e_x and e_y
  parameter Real e_x[:] = normalize(n_x) "={-0.390608730963647,0,-0.920556798516511}";
  parameter Real e_y[:] = normalize(n_y) "={0,-1,0}";
  Orientation T_e = from_nxy(e_x, e_y) "={{-0.390608730963647,0,-0.920556798516511},{0,-1,0},{-0.920556798516511,0,0.390608730963647}}";
  annotation(uses(Modelica(version="3.2.3"))));
end T3201;
```

Again, as reported, the workaround is to always have normalized vectors as inputs to from_nxy and from_nxz.

Since I am not sure if you want to fix the two functions or rather all calls of it, I'll leave the trivial patch to the library officers.

beutlich, 2020-01-07:
I created PR #3294 to directly fix the issue in Modelica.Mechanics.MultiBody.Frames.TransformationMatrices.{from_nxy, from_nxz}.

PR #3294 (2020-01-14): Normalize inputs to from_nxy and from_nxz
Resolves #3201
closed 2020-01-14 by beutlich