modelica/ModelicaStandardLibrary #3218: Improvements of FluxTubes

AHaumer, 2019-11-11:
- Improvement of documentation and reference arrow of ElectroMagneticConverter (should be electric current and magnet voltage)
- Include a hint that coordination between electric and magnetic values is according to a "right-hand screw"
- One ore more simple basic examples that also show reference arrows
- Improvemed icons for shapes, visualizing the shape
- New shape CylinderSectionCircumferentialFlux, i.e. a section (w.r.t. angle) of a hollow cylinder, conducting flux in circumferential direction
- New electric-magnetic-mechanical element, representing a moving coil in an airgap, i.e. a combination of converter and force

I hope I find some time to create different PRs.

I can't find TThomas' colleague Johannes Ziske among the github users ...

AHaumer, 2019-11-11:
@christiankral should'nt we apply the same improvements to QuasiStatic.FluxTubes?

beutlich, 2019-12-15:
> I can't find Thomas Boedrich's colleague Johannes Ziske among the github users ...

I invited @jziske both to MA organizational acoount and MAP-Lib team years ago but he never accepted the invitations. That's the reason why he does not show up in the list of assignees or reviewers.

dietmarw, 2020-01-21:
@AHaumer Can you check if the last two open points are already fixed or can be fixed until tomorrow. Otherwise we should move them to a new ticket for the next version.

AHaumer, 2020-01-22:
Let's shift the last points to the next version ...

dietmarw, 2020-01-22:
@Ahaumer Please make a new ticket with the remaining parts. That way we can include this ticket in the release notes for MSL 4.0.0. Thanks

dietmarw, 2020-01-22:
Further improvements are dealt with by #3359. Thanks @AHaumer

closed 2020-01-22 by dietmarw