modelica/ModelicaStandardLibrary #3254: Utilize absoluteValue=false annotation for relative sensors

mestinso, 2019-12-06:
Given the existence of the absoluteValue=false annotation, I think this Modelica feature should be utilized for relative sensors. Notably the MSL Modelica.Thermal.HeatTransfer.Sensors.RelTemperatureSensor component could directly benefit, since non-absolute scales (Celsius and Fahrenheit scales) are frequently preferred for display units.

A idea for implementation could be:
```
model RelTemperatureSensor "Relative Temperature sensor"
  extends Modelica.Icons.TranslationalSensor;
  Modelica.Thermal.HeatTransfer.Interfaces.HeatPort_a port_a;
  Modelica.Thermal.HeatTransfer.Interfaces.HeatPort_b port_b;
  connector RealOutput_TemperatureDifference = Modelica.Blocks.Interfaces.RealOutput (final unit="K") annotation(absoluteValue=false);
  RealOutput_TemperatureDifference T_rel "Relative temperature as output signal";
equation 
  T_rel = port_a.T - port_b.T;
  0 = port_a.Q_flow;
  0 = port_b.Q_flow;
end RelTemperatureSensor;
```

Alternatively, a new connector interface component ("RealOutputRelative") could be added to Modelica.Blocks.Interfaces if this might be useful elsewhere. Beyond this specific component, I can imagine this at least being useful for subcooling or superheat sensors.

Thoughts?

christiankral, 2019-12-07:
I prefer temperature differences in K as Celsius (and Fahrenheit) units are misleading. 

### Example

Temperatures

- `T1 = 283.15 K = 10 degC`
- `T2 = 263.15 K = -10 degC`

Temperature differences

- `dT1 = T1 - T2 = 20 K` are not equal to `20 degC = 293.15 K` 
- `dT2 = T2 - T1 = -20 K` are not equal to `-20 degC = 253.15 K`

mestinso, 2019-12-07:
While I agree with your point, I don't think your comments are incompatible with my proposal: which is simply to use the annotation for its intended purpose. This way, the user/tool will have improved flexibility if other display units are preferred. This annotation is already being used elsewhere in this library as applicable (via the TemperatureDifference type), so making this change would arguably be more consistent. The only place this annotation isn't being used is in the case of connectors, which show up in sensors such as the one pointed out here.

As is, for any users who have overridden the display unit from Kelvin to Celsius or Fahrenheit, they will see a nonsense result.

christiankral, 2019-12-07:
@HansOlsson is it OK from your side to use `annotation(absoluteValue=false)` in the relative temperature sensor?

HansOlsson, 2019-12-09:
> @HansOlsson is it OK from your side to use annotation(absoluteValue=false) in the relative temperature sensor?

Yes, as that is the intended use. 

I had to check, and since the annotation is on the connector-type it works already in released versions of Dymola.

beutlich, 2019-12-21:
I believe we can't add absoluteValue=false to output y of Modelica.Thermal.FluidHeatFlow.Sensors.RelTemperatureSensor if we do not want to lose inheritance from RelativeSensor.

```mo
model RelTemperatureSensor "Temperature difference sensor"

  extends FluidHeatFlow.Interfaces.RelativeSensor(y(unit="K")
      "Temperature difference as output signal");
// ....
end RelTemperatureSensor;
```

christiankral, 2019-12-21:
I see, this would only work if `absoluteValue` were an attribute that we could modify in an inheritance. Frankly asked @HansOlsson: is this a language shortage?

So the question is what we prefer: 
- Feature of `absoluteValue` in the relative temperature sensors by fully re-writing `Modelica.Thermal.FluidHeatFlow.Sensors.RelTemperatureSensor`
- Modular concept with full inheritance, but without implementing `absoluteValue` 
- Intermediate solution by creating an intermediate layer in `Modelica.Thermal.FluidHeatFlow.Interfaces.Partials`:

```modelica
partial model BaseRelativeSensor "Partial model of relative sensor without signal output"
  extends Modelica.Icons.RotationalSensor;

  parameter FluidHeatFlow.Media.Medium medium=FluidHeatFlow.Media.Medium() "Sensor's medium";
  Interfaces.FlowPort_a flowPort_a(final medium=medium);
  Interfaces.FlowPort_b flowPort_b(final medium=medium);
equation 
  // no mass exchange
  flowPort_a.m_flow = 0;
  flowPort_b.m_flow = 0;
  // no energy exchange
  flowPort_a.H_flow = 0;
  flowPort_b.H_flow = 0;
annotation (...);
end BaseRelativeSensor;

partial model RelativeSensor "Partial model of relative sensor"
  extends BaseRelativeSensor;
  Modelica.Blocks.Interfaces.RealOutput y;
end RelativeSensor;
```

Then the relative sensor could be defined as:

```modelica
model RelTemperatureSensor "Temperature difference sensor"
  extends Interfaces.Partials.BaseRelativeSensor;
  Modelica.Blocks.Interfaces.RealOutput y annotation(absoluteValue = false);
equation 
  ...
end RelTemperatureSensor;
```

beutlich, 2019-12-21:
> Frankly asked @HansOlsson: is this a language shortage?

It is. That's why we have https://github.com/modelica/ModelicaSpecification/pull/2459.

beutlich, 2019-12-21:
> So the question is what we prefer:

Your proposed third option looks good to me.

mestinso, 2019-12-21:
@christiankral why the need to break up Modelica.Thermal.FluidHeatFlow.Interfaces.Partials.RelativeSensor into 3 models? The absoluteValue annotation won't hurt anything for non-temperature cases. Seems like it should be generally applicable (even though the temperature use case is the only one likely to benefit).

beutlich, 2020-01-04:
@HansOlsson Could we also add the absolute value directly to the component, e.g.,

```mo
model RelTemperatureSensor "Relative Temperature sensor"
  extends Modelica.Icons.TranslationalSensor;
  Modelica.Thermal.HeatTransfer.Interfaces.HeatPort_a port_a;
  Modelica.Thermal.HeatTransfer.Interfaces.HeatPort_b port_b;
  Modelica.Blocks.Interfaces.RealOutput T_rel(unit="K", displayUnit="K")
    "Relative temperature as output signal" annotation(absoluteValue=false);
equation 
  T_rel = port_a.T - port_b.T;
  0 = port_a.Q_flow;
  0 = port_b.Q_flow;
end RelTemperatureSensor;
```

? This way we could avoid the dummy type `RealOutput_TemperatureDifference`.

HansOlsson, 2020-01-07:
> @HansOlsson Could we also add the absolute value directly to the component, e.g.,

It is standard Modelica, but it not supported in currently released versions of Dymola and 3D Experience platform.

PR #3293 (2020-01-17): Utilize absoluteValue=false annotation for relative sensors
Resolves #3254.

As confirmed in https://github.com/modelica/ModelicaStandardLibrary/issues/3254#issuecomment-571530727 this is not yet supported by Dymola 2020(x).
closed 2020-01-17 by beutlich