modelica/ModelicaStandardLibrary #3276: Possible Modelica.Media bug

beutlich, 2019-12-18:
**Reported by CTG on 14 Dec 2019 19:53 UTC**

While checking my media models compatibility against ThermoPower, I observed a too high difference in the results obtained in the TestRefrigeranEvaporator test model, using the standard Modelica.Media.R134a.R134a_ph medium and my own medium. I arrived to the conclusion that it was due to the physical properties calculation and checked it. The result I obtain with the Modelica medium seems incorrect regarding density, as you can check with the attached model. The calculation done at 25ºC and 30 bars gives a liquid density result much lower than the saturated one, what seems very strange. I checked the calculation with CoolProp and it seems that the Modelica calculation is erroneous. The calculation at 25ºC is OK till 18 bars.
Can someone check this?

[ModelicaTest5.zip](https://github.com/modelica/ModelicaStandardLibrary/files/3980401/ModelicaTest5.zip)

---

Originally reported at https://openmodelica.org/forum/default-topic/2841-possible-modelica-media-bug

beutlich, 2019-12-18:
**Reported by CTG on 15 Dec 2019 20:02 UTC**

Well, I think I have found the origin of the problem. The medium is giving negative values for  isothermal compressibility, when they should be possitive. Reviewing the code it seems that a term is missing inside the brackets of the kappa calculation.
I hope not to be wrong and that this can help.

---

Originally reported at https://openmodelica.org/forum/default-topic/2841-possible-modelica-media-bug#p9561

beutlich, 2019-12-18:
@wischhusen Can you please analyze? Thank you.

beutlich, 2020-01-13:
@thorade Can you please have a look. I guess he is referring to https://github.com/modelica/ModelicaStandardLibrary/blob/4b560e50519d7b80036550fbbe380f6e1396a00f/Modelica/Media/R134a.mo#L1629-L1630 in Modelica.Media.R134a.R134a_ph.isothermalCompressibility, which even is repeated at https://github.com/modelica/ModelicaStandardLibrary/blob/4b560e50519d7b80036550fbbe380f6e1396a00f/Modelica/Media/R134a.mo#L183

thorade, 2020-01-14:
I believe the analysis is correct and the kappa formula should be:
```
kappa := 1/(state.d*R134aData.data.R_s*state.T*f.delta*(1.0/f.delta + 2.0*f.fdelta + f.delta*f.fdeltadelta)); 
```
or equivalent
```
kappa := 1/(state.d*R134aData.data.R_s*state.T*(1.0 + 2.0*f.delta*f.fdelta + f.delta*f.delta*f.fdeltadelta)); 
```

beutlich, 2020-01-14:
I can confirm that - for the provided test model - `kappa` is now positive. However, `rho` and `cp` are almost) not changed. Thus, it must be something else.

Using `setState_phX(30e5, 245775.707704615)` instead of `setState_pTX(30e5, 298.15)` for `state` changes `rho` and `cp` in similar regions as the expections.

thorade, 2020-01-14:
It might also be that my previous comment was wrong, and the formula is correct. It depends on whether HelmholtzDerivs are split into ideal and residual contribution, or not.
I would assum the issue is somewhere here:
https://github.com/modelica/ModelicaStandardLibrary/blob/af7a2dc8d083ccf7b724c392a0607bdfe84c39b2/Modelica/Media/R134a.mo#L2480-L2552

thorade, 2020-01-14:
New idea: It seems the density found is inside the two-phase region, but dofpt iteration does not protect against solutions in the two-phase region, and the equation of state is evaluated directly.

[image]

beutlich, 2020-01-14:
> It seems the density found is inside the two-phase region, but dofpt iteration does not protect against solutions in the two-phase region, and the equation of state is evaluated directly.

I guess this is the root cause of the problem. I can confirm that rho converges to 969.xxx (6th row) instead of 1190.xxx (4th row).

thorade, 2020-01-14:
I just plotted the residual nderivs.p-p over density and can confirm that it has two roots, at d=969 and at d=1220.

One assert to add could be that nDerivs.pd should never change sign, always be positive, because with increasing pressure density will increase, always. This could help to detect crossing of the spinodal point, but there are possibly more roots inside the two-phase region.

beutlich, 2020-01-14:
Here are the same property tables created by [T3276.mo](https://gist.github.com/beutlich/ccd3505920466e99d0446b1f6d6f2b78):

**setState_phX()** (ref props in parentheses)

| row | T | p | d | h |
|---|---|---|---|---|
| 1 | 374.21 | 4.05911e+06 | 511.899<br>(511.9) | 389639<br>(389639) |
| 2 | 169.85 | 389.563 | 1517.14<br>(1591.11) | 71455.2<br>(71455.2) |
| 3 | 298.15 | 3e+06 | 1220.13<br>(1220.13) | 234687<br>(234687) |
| 4 | 305.987 | 3e+06 | 1190.76<br>(1190.76) | 245776<br>(245776) |
| 5 | 273.15 | 292803 | 1294.76<br>(1294.78) | 200000<br>(200000) |
| 6 | 298.15 | 665380 | 969.978<br>(969.916) | 235741<br>(235741) |

**setState_pTX()** (ref props in parentheses)

| row | T | p | d | h |
|---|---|---|---|---|
| 1 | 374.21 | 4.05911e+06 | 501.213<br>(511.9) | 390932<br>(389639) |
| 2 | 169.85 | 389.563 | n.a.<br>(1591.11) | n.a.<br>(71455.2) |
| 3 | 298.15 | 3e+06 | 969.916<br>(1220.13) | 245776<br>(234687) |
| 4 | 305.987 | 3e+06 | 938.83<br>(1190.76) | 256324<br>(245776) |
| 5 | 273.15 | 292803 | n.a.<br>(1294.78) | n.a.<br>(200000) |
| 6 | 298.15 | 665380 | n.a.<br>(969.916) | n.a.<br>(235741) |

Whereas setState_phX is a perfect model (with slight exception of row #⁠2 where p=PMIN and T=TMIN), setState_pTX is off in row #⁠4 and #⁠5 for both density and enthalpy.

If setState_pTX cannot be fixed, it should at least assert and not return wrong results.

beutlich, 2020-01-15:
I marked the incorrect values by :small_red_triangle: in the above table for better visualibility.

PR #3491 (2020-03-09): maint/3.2.3: Fix Modelica.Media.R134a.R134a_ph.setState_pTX (only applicable for one-phase)
Back-port #3463 to also fix #3276 in maint/3.2.3.
closed 2020-03-02 by beutlich