modelica/ModelicaStandardLibrary #3278: The gh-pages branch cannot be checked out cleanly on Windows

HansOlsson, 2019-12-20:
Merely **checking** out gh-pages on Windows leads to more than a dozen of "Changes not staged for commit". The reason is that there are multiple files that only differ by case, as can be seen by comparing:

https://doc.modelica.org/Modelica%203.2.3/Resources/helpWSM/Modelica/Modelica.Electrical.Spice3.Internal.BJT.html
https://doc.modelica.org/Modelica%203.2.3/Resources/helpWSM/Modelica/Modelica.Electrical.Spice3.Internal.Bjt.html

dietmarw, 2019-12-20:
I guess this can help you here: https://www.windowscentral.com/how-enable-ntfs-treat-folders-case-sensitive-windows-10

beutlich, 2019-12-20:
You need to install WSL for that, see https://stackoverflow.com/questions/50839623/fsutil-the-request-is-not-supported-after-setcasesensitiveinfo

HansOlsson, 2019-12-20:
Well, a simpler work-around was to just ignore it and not stage all changes.
But I expect that it will confuse others as well.

beutlich, 2019-12-21:
> 
> 
> I guess this can help you here: https://www.windowscentral.com/how-enable-ntfs-treat-folders-case-sensitive-windows-10

Seems to work once you have WSL installed on Win10.

[image]

closed 2019-12-21 by beutlich