modelica/ModelicaStandardLibrary #3283: Modelica.Electrical.Analog.Examples.SimpleTriacCircuit flawed

JoeRiel, 2019-12-21:
The test circuit, as is, fails in MapleSim. It also fails in Dymola if R1 is changed. The problem is pulseCurrent, used to trigger the gate. When the sine wave load is negative, the trigger current must flow through the load, which is inductive. Pushing a pulse current, with a rise-time of 0, through an inductive load is a non-physical action. The simplest fix I could come up with is to replace the pulse current source with a signal current source driven by a pulse driver run through a 1-pole filter to achieve non-zero edges.

christiankral, 2019-12-22:
Alternatively we could replace the `PulseCurrent` by the `TrapezoidCurrent` source.

AHaumer, 2019-12-22:
@JoeRiel please check whether PR #3284 works

PR #3284 (2019-12-23): Fix IdealTriacCircuit example
Should resolve #3283
closed 2019-12-23 by beutlich