modelica/ModelicaStandardLibrary #3295: Unit of permeability

AHaumer, 2020-01-08:
- SIunits.Permeability is defined with unit "H/m".
- Constants.mu_0 is defined as real with unit "N/A2".
- Both units are correct, we could also use "V.s/(A.m)".

Two questions:

1. Do all tools figure out that N/A2 = H/m = V.s/(A.m) ?
2. Shouldn't Constants.mu_0 define `SIunits.Permeability mu_0` instead of `Real mu_0(unit="N/A2")`?

dietmarw, 2020-01-09:
1. My personal preference would be to use `V.s/(A.m)` as they are SI units and not derived. 
2. Yes I agree.

HansOlsson, 2020-01-09:
> Two questions:
> 
> Do all tools figure out that N/A2 = H/m = V.s/(A.m) ?

Yes, if tools are able to do any reasonable unit-checking (which I believe most are) that is trivial.

https://en.wikipedia.org/wiki/Vacuum_permeability uses both H/m and N/A2, and give reasonable explanations for both, and also indicates a larger issue - the constant no longer has a fixed simple value, but now seems to be 4π × 1.000 000 000 82 (20) 1e-7.

I can make a PR. (Just changing the value, not adding the indirect definition, since that definition requires yet another measured constant that I don't see any libraries using.)

> Shouldn't Constants.mu_0 define SIunits.Permeability mu_0 instead of Real mu_0(unit="N/A2")?

That would work as well, but it depends on how dependent we want to make the libraries. 
However, since we use it for the other constants it would make sense.

HansOlsson, 2020-01-09:
> Two questions:
> Do all tools figure out that N/A2 = H/m = V.s/(A.m) ?
>
> Yes, if tools are able to do any reasonable unit-checking (which I believe most are) that is trivial. [...]

Have made PR with the correct value, now it is just a matter of fixing the bad value on Wikipedia!

beutlich, 2020-01-09:
I guess you mean:

[image]

HansOlsson, 2020-01-09:
> I guess you mean:

I meantchanging it from 4π × 1.000 000 000 82 (20) 1e-7 to 4π × 1.000 000 000 55 (15) 1e-7; as given in https://physics.nist.gov/cuu/pdf/all.pdf

https://en.wikipedia.org/w/index.php?title=Vacuum_permeability&type=revision&diff=934922863&oldid=921057044

HansOlsson, 2020-01-09:
Reading more I'm not sure if we should change the type of the constant, as there are some indications that mu_0 isn't seen as "permeability of vacuum" anymore, but as a "magnetic constant".

https://en.wikipedia.org/wiki/Vacuum_permeability

I don't know if there is a deep reason for that (and if so what it is), but assuming there is it seems odd to use the type "Permeability". On the other hand https://physics.nist.gov/cuu/Constants/codata.pdf uses "magnetic constant" and mu/mu_0 as a relative permeability - indicating that mu_0 is also a permeability.

AHaumer, 2020-01-12:
In fact mu_0 is a permeability ...

PR #3811 (2021-05-22): Change unit of permeability to V.s/(A.m)
Fixes #3295
commit 3b2325cf (2021-05-22): Fixes #3295 by changing the unit to V.s/(A.m) and using Permeability
closed 2021-05-22 by AHaumer