modelica/ModelicaStandardLibrary #3300: LeakageWithCoefficent should have a port for reluctance input

JoeRiel, 2020-01-10:
The model Modelica.Magnetic.FluxTubes.Basic.LeakageWithCoefficient has an input signal, R_mUsefulTot, that provides the total reluctance of the path in which it is connected. To make this part usable in a GUI, that input should be a port to which a real signal can be connected. 

Actually, it's not clear to me why this quantity is an input and not a parameter. Are there examples of a model where it varies?  The doc-string for the component ("... not for dynamic simulation of actuators") suggests it isn't intended to be used in that application.

AHaumer, 2020-01-12:
I'm not sure whether `input` should be a `parameter`, I'll ask the authors about their intentions. 
"... not for dynamic simulation of actuators": Please read the full doc, the authors wrote:
" ...This would generate a reluctance force in the leakage element that is not accounted for properly."
I interpret this thay way that if you don'ct care about the reluctance force you can use this leakage element in parallel to the "usefull" element, assuming that the coupling coefficient remains constant which in turns leads to a change of the leakage reluctance if "usefull" reluctanec changes. 
Following the idea, `R_mUsefulTot` should be an input rather than a parameter.

AHaumer, 2020-01-12:
I've created a PR, showing the usage in the Example `BasicExamples.QuadraticCoreAirgap`

christiankral, 2020-02-15:
Re-open due to https://github.com/modelica/ModelicaStandardLibrary/pull/3424#issuecomment-586562266

dietmarw, 2020-02-18:
Fixed again by #3424

dietmarw, 2020-02-18:
Probably wait for #3431 first...

PR #3307 (2020-01-14): Fix #3300 LeakageWithCoefficient
fixes #3300
commit 1ccaaccd (2020-01-14): fix #3300 LeakageWithCoefficient
commit 4d0df91f (2020-01-14): Merge pull request #3307 from AHaumer/LeakageWithCoefficient

Fix #3300 LeakageWithCoefficient
PR #3431 (2020-02-20): Fix unconsidered change of parameter c_usefulFlux from default to start
Fixes #3300
Refs #3424

@beutlich: I suppose there is some scheme on how the conversion script commands are sorted and arranged in `ConvertModelica_from_3.2.3_to_4.0.0.mos`, respectively. I hope you agree with on where I put the code.

I also added a test model to `ModelicaTestConversion4.mo` to test both cases: with and without default parameter. Both cases are converted OK.
closed 2020-02-20 by dietmarw