modelica/ModelicaStandardLibrary #3305: Naming of analog sources

christiankral, 2020-01-12:
The package `Modelica.Electrical.Analog.Sources` contains models which naming we may want to re-consider:

- `ExpSineVoltage`
- `ExponentialsVoltage`
- `ExpSineCurrent`
- `ExponentialsCurrent`

Two questions arise to me (certainly the same questions for voltage and current sources):

1. Is `ExponentialsVoltage` correct (which sounds plural to me) or shall it be ``ExponentialVoltage`? 
2. Shall we change `ExpSineCurrent` to `ExponentialSineCurrent` (or `ExponentialsSineCurrent`?) in order to not use abbreviations. The length of the class name shall not be an issue here, as we have much longer names in this package, too:

[image]

In addition to the library officers, hopefully a native English speaking person wants to contribute here.

AHaumer, 2020-01-12:
IMHO `ExponentialSineVoltage` is correct since the amplitude of the sine is exponentially damped, and yes it is a good idea to get rid of these abbreviations. `ExponentialVoltage`  is misleading, I'd expect a voltage varyimg exponentially with respect to time. `ExponentialsSineVoltage` (plural) is wrong since we have one exponential (i.e. the variation of amplitude). We shouln't forget `Modelica.Blocks.Sources.ExpSine`!
`ExponentialsVoltage` (plural) sound correct to me since we have a rising and a falling exponential with different time constants.
Of course all said applies to Crrent sources, too.

beutlich, 2020-03-02:
Removing milestone.