modelica/ModelicaStandardLibrary #3316: Messed up section numbers for MoMoLib_VerificationResults_XRG.pdf

beutlich, 2020-01-14:
When trying to understand the R134a verification of [MoMoLib](http://www.xrg-simulation.de/de/forschung/momolib) vs. RefProp for #3276 by reading [MoMoLib_VerificationResults_XRG.pdf](https://github.com/modelica/ModelicaStandardLibrary/blob/8f3bf84ba216c03f0f1e099e8b8fa24961591957/Modelica/Resources/Documentation/Media/MoMoLib_VerificationResults_XRG.pdf), I noticed that the heading numbers are messed up for subsections of section 2.1.

If we fix it, we should also emphasize, that this then is an updated document and no longer the original document handed in for Clean Sky.

wischhusen, 2020-02-05:
Resolved. Pushed a new pdf to my fork.

beutlich, 2020-02-05:
The PDF you added by https://github.com/wischhusen/Modelica-1/commit/e33c605ffd6d98780cca4fb91022e6833dcd5700 only contains 17 pages whereas the previous version contained 24 pages. Can you please create a full PDF that still contains the sections 3 and 4.

wischhusen, 2020-02-06:
Sorry, I added the missing parts.

PR #3395 (2020-02-07): Fix section numbers in MoMoLib_VerificationResults_XRG.pdf
Closes #3316.

@wischhusen I downgraded the PDF version to 1.4 (was 1.7 in your commit).
closed 2020-02-07 by beutlich