modelica/ModelicaStandardLibrary #3320: Erroneous two-phase calculation of function setState_psX for R134a medium

beutlich, 2020-01-14:
I am afraid the issue reported in #3163 for Modelica.Media.R134a.R134a_ph.setState_dTX (which was fixed in #3275) also is apparent with Modelica.Media.R134a.R134a_ph.setState_psX in two-phase region.

Check the setState_psX table generated by [T3320.mo](https://gist.github.com/beutlich/fd5aaea3a770aa7668c3ebb96592ad19):

**setState_phX()** (ref props in parentheses)

| row | T | p | d | h |
|---|---|---|---|---|
| 1 | 374.21<br>(374.21) | 4.05911e+06<br>(4.05911e+06) | 511.899<br>(511.9) | 389639<br>(389639) |
| 2 | 169.85<br>(169.85) | 389.563<br>(389.563) | 1517.14<br>(1591.11) | 71455.2<br>(71455.2) |
| 3 | 298.15<br>(298.15) | 3e+06<br>(3e+06) | 1220.13<br>(1220.13) | 234687<br>(234687) |
| 4 | 305.987<br>(305.987) | 3e+06<br>(3e+06) | 1190.76<br>(1190.76) | 245776<br>(245776) |
| 5 | 273.15<br>(273.15) | 292803<br>(292803) | 1294.76<br>(1294.78) | 200000<br>(200000) |
| 6 | 298.15<br>(298.15) | 665380<br>(665380) | 969.978<br>(969.916) | 235741<br>(235741) |

**setState_dTX()** (ref props in parentheses)

| row | T | p | d | h |
|---|---|---|---|---|
| 1 | 374.21<br>(374.21) | 4.05911e+06<br>(4.05911e+06) | 511.9<br>(511.9) | 389639<br>(389639) |
| 2 | 169.85<br>(169.85) | 2527.38<br>(389.563) | 1591.11<br>(1591.11) | 71456<br>(71455.2) |
| 3 | 298.15<br>(298.15) | 3.00047e+06<br>(3e+06) | 1220.13<br>(1220.13) | 234687<br>(234687) |
| 4 | 305.987<br>(305.987) | 3.00047e+06<br>(3e+06) | 1190.76<br>(1190.76) | 245776<br>(245776) |
| 5 | 273.15<br>(273.15) | 293558<br>(292803) | 1294.78<br>(1294.78) | 200000<br>(200000) |
| 6 | 298.15<br>(298.15) | 665381<br>(665380) | 969.916<br>(969.916) | 235741<br>(235741) |

**setState_psX()** (ref props in parentheses)

| row | T | p | d | h |
|---|---|---|---|---|
| 1 | 374.21<br>(374.21) | 4.05911e+06<br>(4.05911e+06) | 511.897<br>(511.9) | 389639<br>(389639) |
| 2 | 293.15<br>(169.85) | 389.563<br>(389.563) | 4<br>(1591.11) | 71455.2<br>(71455.2) |
| 3 | 298.149<br>(298.15) | 3e+06<br>(3e+06) | 1220.13<br>(1220.13) | 234686<br>(234687) |
| 4 | 305.986<br>(305.987) | 3e+06<br>(3e+06) | 1190.76<br>(1190.76) | 245775<br>(245776) |
| 5 | 293.15<br>(273.15) | 292803<br>(292803) | 4<br>(1294.78) | 200000<br>(200000) |
| 6 | 293.15<br>(298.15) | 665380<br>(665380) | 4<br>(969.916) | 235741<br>(235741) |

**setState_pTX()** (ref props in parentheses)

| row | T | p | d | h |
|---|---|---|---|---|
| 1 | 374.21<br>(374.21) | 4.05911e+06<br>(4.05911e+06) | 501.2<br>(511.9) | 390934<br>(389639) |
| 2 | 169.85<br>(169.85) | 389.563<br>(389.563) | n.a.<br>(1591.11) | n.a.<br>(71455.2) |
| 3 | 298.15<br>(298.15) | 3e+06<br>(3e+06) | 1220.13<br>(1220.13) | 234687<br>(234687) |
| 4 | 305.987<br>(305.987) | 3e+06<br>(3e+06) | 1190.76<br>(1190.76) | 245776<br>(245776) |
| 5 | 273.15<br>(273.15) | 292803<br>(292803) | n.a.<br>(1294.78) | n.a.<br>(200000) |
| 6 | 298.15<br>(298.15) | 665380<br>(665380) | n.a.<br>(969.916) | n.a.<br>(235741) |

beutlich, 2020-01-15:
I marked the incorrect values by :small_red_triangle: in the above table for better visualibility.

wischhusen, 2020-02-05:
Resolved as far as I can see. Outputs for d and T were missing in setState_psX in two-phase mode.

beutlich, 2020-02-05:
> 
> 
> Resolved as far as I can see.

What do you mean by "Resolved"? Do you have a patch for it?

wischhusen, 2020-02-05:
Yes I pushed into my fork

PR #3392 (2020-02-07): Add missing property calculation for d and T in Modelica.Media.R134a.R134a_ph.setState_psX
Additionally, add the phase calculation for the thermodynamic state.

Closes #3320.
PR #3397 (2020-02-11): maint/3.2.3: Add missing property calculation for d and T in Modelica.Media.R134a.R134a_ph.setState_psX
Back-port #3392 to fix #3320 also in maint/3.2.3 (since labeled as bug).
closed 2020-02-07 by beutlich