modelica/ModelicaStandardLibrary #3339: Not used images

tobolar, 2020-01-17:
There are image files in the following directories not used elswhere.
- Modelica\Resources\Images\Fluid\Components
- Modelica\Resources\Images\Fluid\Dissipation\heatTransfer
- Modelica\Resources\Images\Fluid\Dissipation\pressureLoss

Note: Some of the images are used for Fluid icons.

casella, 2020-02-12:
I'm sorry but I really have no time to worry about this. Anything you did is fine for me

dietmarw, 2020-02-12:
@casella Could you please in your role as library officer take the time to come to a decision as to what to do with those. It is not critical for the MSL 4.0.0 release but simply saying it's somebody else's problem is not a solution :frowning: 
At least come up with a plan on when and how to deal with this.

casella, 2020-02-13:
> @casella Could you please in your role as library officer take the time to come to a decision as to what to do with those.

IMHO this issue has a very, very low priority. I do not have time now to fully understand what the situation is, in order to take an informed decision on any action. I prefer to spend my limited time on more important issues, e.g. what to do with homotopy or how to fix models that produce non-physical results.

Hence, my decision is: defer this issue until the next release, unless anybody wants to pick this up (e.g. @tobolar), in which case I'm happy to completely delegate it.

wischhusen, 2020-02-17:
I will have a look. Please give me this week for checking.

wischhusen, 2020-02-17:
@beutlich I think I removed everything obsolete.

dietmarw, 2020-02-17:
@wischhusen Where did you do that? You should send a PR so we can merge the changes.

wischhusen, 2020-02-17:
in my fork

dietmarw, 2020-02-17:
Could you please create a PR and reference this ticket (e.g., "ref: #3339") in the comment. Thanks.

wischhusen, 2020-02-17:
no i can't.

dietmarw, 2020-02-17:
Why not? What is the problem?

wischhusen, 2020-02-17:
[image]

dietmarw, 2020-02-17:
Try this link: https://github.com/wischhusen/Modelica-1/pull/new/master

dietmarw, 2020-02-18:
A cleaned up PR is available here: #3429

beutlich, 2020-02-22:
@dietmarw Can this be closed after merging #3429?

dietmarw, 2020-02-22:
Fixed by #3429

closed 2020-02-22 by dietmarw