modelica/ModelicaStandardLibrary #3341: Broken class links in documentation need fixing.

dietmarw, 2020-01-17:
By running the OM document generator I found a couple of missing class links in the documentation which need to be handled manually (and could not be fixed by #3340). Often classes where renamed or relocated and it was forgotten to update the links in the documentation. Here the error report from the OM documentation generator:

```
[Checking file Modelica.Electrical.Batteries.BaseClasses.BaseStackWithSensors.html]:
Not found: Modelica.Electrical.Batteries.StacksWithMeasurement.Cell.html
[Checking file Modelica.Electrical.Batteries.BatteryStacks.CellStack.html]:
Not found: Modelica.Electrical.Batteries.ParameterRecords.RCData.html
[Checking file Modelica.Electrical.Batteries.BatteryStacksWithSensors.Stack.html]:
Not found: Modelica.Electrical.Batteries.StacksWithMeasurement.Cell.html
[Checking file Modelica.Electrical.Batteries.BatteryStacksWithSensors.StackRC.html]:
Not found: Modelica.Electrical.Batteries.StacksWithMeasurement.CellRC.html
[Checking file Modelica.Electrical.Batteries.Examples.html]:
Not found: Modelica.Electrical.Batteries.SimpleBatteries.html
[Checking file Modelica.Electrical.Batteries.ParameterRecords.TransientData.CellData.html]:
Not found: Modelica.Electrical.Batteries.ParameterRecords.RCData.html
[Checking file Modelica.Electrical.Batteries.UsersGuide.Parameterization.html]:
Not found: Modelica.Electrical.Batteries.ParameterRecords.RCData.html
[Checking file Modelica.Electrical.PowerConverters.UsersGuide.ACACConcept.html]:
Not found: Modelica.Electrical.PowerConverters.DCDC.Control.Signal2mPulse.html
[Checking file Modelica.Magnetic.FluxTubes.Shapes.FixedShape.html]:
Not found: Modelica.Magnetic.FluxTubes.Interfaces.PartialFixedShape.html
[Checking file Modelica.Magnetic.FluxTubes.Shapes.Force.html]:
Not found: Modelica.Magnetic.FluxTubes.Interfaces.PartialForce.html
[Checking file Modelica.Magnetic.FluxTubes.Shapes.Leakage.html]:
Not found: Modelica.Magnetic.FluxTubes.Interfaces.PartialForce.html
[Checking file Modelica.Magnetic.FluxTubes.UsersGuide.ReleaseNotes.html]:
Not found: Modelica.Magnetic.FluxTubes.Interfaces.PartialFixedShape.html
[Checking file Modelica.Magnetic.FluxTubes.UsersGuide.ReluctanceForceCalculation.html]:
Not found: Modelica.Magnetic.FluxTubes.Interfaces.PartialForce.html
[Checking file Modelica.Magnetic.FundamentalWave.UsersGuide.ReleaseNotes.html]:
Not found: Modelica.Magnetic.FundamentalWave.Interfaces.PartialTwoPort.html
[Checking file Modelica.Magnetic.FundamentalWave.UsersGuide.ReleaseNotes.html]:
Not found: Modelica.Magnetic.FundamentalWave.Interfaces.PartialTwoPortElementary.html
[Checking file Modelica.Magnetic.QuasiStatic.FluxTubes.Interfaces.RelativeSensor.html]:
Not found: Modelica.Magnetic.QuasiStatic.FluxTubes.Interfaces.PartialTwoPorts.html
[Checking file Modelica.Magnetic.QuasiStatic.FundamentalWave.UsersGuide.ReleaseNotes.html]:
Not found: Modelica.Magnetic.QuasiStatic.FundamentalWave.Interfaces.PartialTwoPortExtended.html
[Checking file Modelica.Mechanics.Rotational.Interfaces.InternalSupport.html]:
Not found: Modelica.Mechanics.Rotational.Interfaces.PartialElementaryOneFlangeAndSupport.html
[Checking file Modelica.Mechanics.Rotational.Interfaces.InternalSupport.html]:
Not found: Modelica.Mechanics.Rotational.Interfaces.PartialElementaryTwoFlangesAndSupport.html
[Checking file Modelica.Mechanics.Translational.Interfaces.InternalSupport.html]:
Not found: Modelica.Mechanics.Translational.Interfaces.PartialElementaryOneFlangeAndSupport.html
[Checking file Modelica.Mechanics.Translational.Interfaces.InternalSupport.html]:
Not found: Modelica.Mechanics.Translational.Interfaces.PartialElementaryTwoFlangesAndSupport.html
```

christiankral, 2020-02-14:
OK, I will create a PR

beutlich, 2020-02-22:
@dietmarw Is this issue resolved finally?

commit 8eac1190 (2020-02-22): Finally fixes #3341 by correcting last broken link.
closed 2020-02-22 by dietmarw