modelica/ModelicaStandardLibrary #3342: Bug in Modelica.Electrical.QuasiStationary.Multiphase.Basic.MutualInductor?

svorkoetter, 2020-01-17:
The model `Modelica.Electrical.QuasiStationary.Multiphase.Basic.MutualInductor` is as follows:

    model MutualInductor "Linear mutual inductor"
      extends Modelica.Electrical.QuasiStationary.MultiPhase.Interfaces.OnePort;
      import Modelica.ComplexMath.j;
      parameter Real epsilon=1e-9 "Relative accuracy tolerance of matrix symmetr
      parameter Modelica.SIunits.Inductance L[m, m] "Mutual inductance matrix";
    initial equation
      if abs(Modelica.Math.Matrices.det(L)) < epsilon then
        Modelica.Utilities.Streams.print("Warning: mutual inductance matrix sing
      end if;
    equation
      assert(sum(abs(L - transpose(L))) < epsilon*sum(abs(L)),"Mutual inductance
      for j in 1:m loop
        v[j] = sum(j*omega*L[j, k]*i[k] for k in 1:m);
      end for;
    end MutualInductor;

I'm a bit concerned about the for-loop. Notice that `j` appears twice in the `sum` expression. I could be wrong (I'm not an EE), but I think the first `j` is meant to be the imaginary unit imported by the `import` clause. Of course, a tool will treat it as the loop index `j`.

svorkoetter, 2020-01-17:
Actually, reading the documentation annotation, it's clear that the first `j` is meant to be the imaginary unit. Thus, this model is definitely broken.

beutlich, 2020-01-17:
I changed the milestone to v4.0.0. If it is a bug, I'll care for the back-port to v3.2.3 afterwards.

PR #3343 (2020-01-17): Fix bug in quasi static MutualInductor
Resolves #3342.
PR #3344 (2020-01-17): maint/3.2.3: Fix bug in quasi-static MutualInductor
Back-port #3343 to fix #3342 in maint/3.2.3.
closed 2020-01-17 by beutlich