modelica/ModelicaStandardLibrary #3349: Immoderate size of simulation result data for Rectifier6pulseFFT and Rectifier12pulseFFT

beutlich, 2020-01-19:
Dymola simulation of Rectifier6pulseFFT creates a MAT file of size 3.75 GB. This is for 5662 variables of dimension 101802. I am not sure if this is the desired purpose of the example model.

Similar for Rectifier12pulseFFT where the size of the simulation result data even is greater than 4 GB.

Maybe we could spend some HideResult=true annotations to reduce the size of the simulation result data.

HansOlsson, 2020-01-20:
As far as I understand that simulation result isn't important at all.

The primary output is the file "rectifier6pulseFFTresult.mat" (and similarly for the other example), which is about 29kB. 

However, you only get the really large result when storing protected variables, as far as I know so I don't see it as a major issue for users. (It would be fairly simple to reduce the simulation size by a factor 3 by making abs and arg function-local variables.)

AHaumer, 2020-01-20:
Those two examples are derived from `Electrical.Machines.Examples.Transformers.Rectifier6pulse` (resp. 12 pulse). To see the trajectory of currents, a rather short simulation time is neeeded, but with a high resolution (Interval length). 
Using the FFT-block, we have to ensure that "that enough points can be sampled before the simulation ends: startTime + (ns - 1)*samplePeriod". I have to re-check the simulation time. 
This depends on maximum frequency and frequency resolution. Even if we try to set a longer interval (which in principle leads to aliasing of the trajectories), the FFT-block samples its input producing events. I suppose that all results normally are stored at events. I'll have to perform some tests.

AHaumer, 2020-01-20:
We can reduce the simulation time, this reduces the size of the result file. 
I wouldn't increase the interval, otherwise we get trajectories with aliasing. 
I'll create a PR.

PR #3352 (2020-01-20): Reduce result file size of FFT examples (#3349)
I reduced stop time, adding some hints in the documentation about maximum frequency, frequency resolution, sample time and sample period. 
If needed, we could try some HideResult=true annotations additionally.

Resolves #3349.
closed 2020-01-20 by beutlich