modelica/ModelicaStandardLibrary #3378: TestWallFriction fails with pedantic check

dietmarw, 2020-01-30:
Whilst testing the major SI unit PR I checked the MSL (4.0.0-alpha.1 (almost)) in Dymola in pedantic mode and got the following error on `Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.TestWallFriction`:

```
Check of Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.TestWallFriction:

Component type specifier MassFlowRate is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 13
Component context: port_a.m_flow
Component declared as MassFlowRate m_flow in Modelica.Media.Interfaces.PartialMedium

Component type specifier MassFlowRate is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 13
Component context: port_b.m_flow
Component declared as MassFlowRate m_flow in Modelica.Media.Interfaces.PartialMedium

Component type specifier MassFlowRate is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 289
Component context: m_flow_start
Component declared as MassFlowRate m_flow_start in Modelica.Media.Interfaces.PartialMedium

Component type specifier MassFlowRate is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 293
Component context: m_flow_small
Component declared as MassFlowRate m_flow_small in Modelica.Media.Interfaces.PartialMedium

Component type specifier MassFlowRate is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 306
Component context: m_flow
Component declared as MassFlowRate m_flow in Modelica.Media.Interfaces.PartialMedium

Component type specifier ThermodynamicState is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 332
Component context: state_a
Component declared as ThermodynamicState state_a in Modelica.Media.Interfaces.PartialMedium

Component type specifier ThermodynamicState is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 333
Component context: state_b
Component declared as ThermodynamicState state_b in Modelica.Media.Interfaces.PartialMedium

Component type specifier MassFlowRate is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 714
Component context: m_flow
Component declared as MassFlowRate m_flow in Modelica.Media.Interfaces.PartialMedium

Component type specifier MassFlowRate is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Interfaces.mo, line 726
Component context: m_flow_start
Component declared as MassFlowRate m_flow_start in Modelica.Media.Interfaces.PartialMedium

Component type specifier ThermodynamicState is from a partial package. That is not correct.
File: /home/dietmarw/.workspace/gh-modelica/Modelica/Modelica/Fluid/Fittings.mo, line 1570
Component context: state_nominal
Component declared as ThermodynamicState state_nominal in Modelica.Media.Interfaces.PartialMedium

ERRORS have been issued.
```

beutlich, 2020-01-30:
This issue is independent of #3379 and already can observed with MSL v3.2.3 in dymola 2020.

HansOlsson, 2020-01-31:
That error message when checking the specific model is a bit excessive (and not exclusive to this model as far as I can tell). It does not occur when checking the entire package.

It should say "It must be redeclared before being used." (as similar messages do); and the underlying problem is that the types are in the partial package Medium that must be redeclared.

(Note: Just so that no-one tries to "correct" this by making the model partial: you can currently drag in this model and redeclare the Medium - and then it just works; similarly as if a parameter lacked a value.)

wischhusen, 2020-01-31:
I do not understand the purpose of the model. What kind of tests were intended? Even if a suitable fluid model is chosen failures will remain due to the missing inner System model. I would rather say let us remove the model ...

beutlich, 2020-01-31:
It is not used anywhere inside MSL or ModelicaTest.

HansOlsson, 2020-01-31:
> 
> 
> It is not used anywhere inside MSL or ModelicaTest.

It is used in `ModelicaTest.Fluid.TestComponents.Fittings.TestWallFriction` - that compares different pipes and also this one. Thus one possibility would be to move it to ModelicaTest, as it seems it isn't intended for actual use (according to description).

beutlich, 2020-01-31:
Ah, sorry. I mixed it up with Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.BaseModel which is nowhere used.

PR #3383 (2020-02-03): Move test-only models TestWallFriction and TestWallFrictionAndGravity to ModelicaTest
Move the test-only models

- [x] Modelica.Fluid.Fittings.BaseClasses.QuadraticTurbulent.TestWallFriction
- [x] Modelica.Fluid.Pipes.BaseClasses.WallFriction.TestWallFrictionAndGravity

to ModelicaTest.Fluid.BaseClasses.

Closes #3378.
closed 2020-02-03 by beutlich