modelica/ModelicaStandardLibrary #3384: Failed conversion of convertElement from R to R_s

beutlich, 2020-02-02:
The following test package was extracted from the medium Buildings.Media.Air of the [Buildings library](https://github.com/lbl-srg/modelica-buildings):

```mo
package ConvTestPackage
  partial package ConvTest
    extends Modelica.Media.Interfaces.PartialCondensingGases;
    redeclare replaceable partial model extends BaseProperties
    equation
      R = Modelica.Constants.R/MM;
    end BaseProperties;
  end ConvTest;
  annotation(uses(Modelica(version="3.2.3")));
end ConvTestPackage;
```

It successfully checks with MSL v3.2.3, but the conversion fails if updated to MSL v4.0.0.

I added the missing lines in ConvertModelica_from_3.2.3_to_4.0.0.mos

```
convertElement({"Modelica.Media.Interfaces.PartialMixtureMedium.BaseProperties",
                "Modelica.Media.Interfaces.PartialCondensingGases.BaseProperties",
                "Modelica.Media.Interfaces.PartialRealCondensingGases.BaseProperties"},
                "R", "R_s")
```

but still the conversion does not work out.

@HansOlsson Is this a conversion issue of #3190, a tool issue of Dymola 2020 or even specification issue of the conversion scripts? Can you please analyze. Thanks.

beutlich, 2020-02-07:
Same issue arises when converting the [TRANSFORM-Library](https://github.com/ORNL-Modelica/TRANSFORM-Library).

HansOlsson, 2020-02-10:
> 
> 
> Same issue arises when converting the [TRANSFORM-Library](https://github.com/ORNL-Modelica/TRANSFORM-Library).

That one seems impossible to handle automatically without any changes of the library, since
TRANSFORM.Media.Interfaces.Fluids.PartialMedium.BaseProperties
contains a duplicate of the entire class including "R", and there is no real direct link to the MSL-variant.

HansOlsson, 2020-02-10:
> 
> 
> The following test package was extracted from the medium Buildings.Media.Air of the [Buildings library](https://github.com/lbl-srg/modelica-buildings):
> 
> ```modelica
> package ConvTestPackage
>   partial package ConvTest
>     extends Modelica.Media.Interfaces.PartialCondensingGases;
>     redeclare replaceable partial model extends BaseProperties
>     equation
>       R = Modelica.Constants.R/MM;
>     end BaseProperties;
>   end ConvTest;
>   annotation(uses(Modelica(version="3.2.3")));
> end ConvTestPackage;
> ```
> 
> It successfully checks with MSL v3.2.3, but the conversion fails if updated to MSL v4.0.0.
> 
> I added the missing lines in ConvertModelica_from_3.2.3_to_4.0.0.mos
> 
> ```
> convertElement({"Modelica.Media.Interfaces.PartialMixtureMedium.BaseProperties",
>                 "Modelica.Media.Interfaces.PartialCondensingGases.BaseProperties",
>                 "Modelica.Media.Interfaces.PartialRealCondensingGases.BaseProperties"},
>                 "R", "R_s")
> ```
> 
> but still the conversion does not work out.
> 
> @HansOlsson Is this a conversion issue of #3190, a tool issue of Dymola 2020 or even specification issue of the conversion scripts? Can you please analyze. Thanks.

It seems like it is only a tool-issue for this case, and it will be improved for Dymola 2021.
However, as previously noted I don't see how we could handle the TRANSFORM-library.

beutlich, 2020-02-10:
Thanks for the feeedback. Do we need to add the missing conversion rules

```
convertElement({"Modelica.Media.Interfaces.PartialMixtureMedium.BaseProperties",
                "Modelica.Media.Interfaces.PartialCondensingGases.BaseProperties",
                "Modelica.Media.Interfaces.PartialRealCondensingGases.BaseProperties"},
                "R", "R_s")
```

assuming it was not correctly detected previously by your Dymola dev version for PR #3195?

HansOlsson, 2020-02-11:
> Thanks for the feeedback. Do we need to add the missing conversion rules
>
> ```
> convertElement({"Modelica.Media.Interfaces.PartialMixtureMedium.BaseProperties", [...]

Yes. They need to be added.

commit 00d296b7 (2020-02-11): fix #3384, refs #3190: Add missing conversion rules
closed 2020-02-11 by beutlich