modelica/ModelicaStandardLibrary #3387: Use real enumeration in e.g. Modelica.Mechanics.Rotational.Interfaces.PartialFriction!

HansOlsson, 2020-02-03:
The following constants and their use could be replaced by an enumeration:

 ```
 constant Integer Unknown=3 "Value of mode is not known";
  constant Integer Free=2 "Element is not active";
  constant Integer Forward=1 "w_relfric > 0 (forward sliding)";
  constant Integer Stuck=0
    "w_relfric = 0 (forward sliding, locked or backward sliding)";
  constant Integer Backward=-1 "w_relfric < 0 (backward sliding)";
  Integer mode(
    final min=Backward,
    final max=Unknown,
    start=Unknown,
    fixed=true)
    "Mode of friction (-1: backward sliding, 0: stuck, 1: forward sliding, 2: inactive, 3: unknown)";
```
The constants seem to occur in 
Modelica.Mechanics.Rotational.Components.LossyGear
Modelica.Mechanics.Rotational.Interfaces.PartialFriction
Modelica.Mechanics.Translational.Interfaces.PartialFriction
Modelica.Mechanics.Translational.Components.PartialFrictionWithStop
Modelica.Mechanics.Translational.Components.MassWithStopAndFriction

That was what you meant @DagBruck ?

tobolar, 2023-10-25:
Would a conversion script "type conversion from Integer to enumeration" work seamlessly?

And, the range of integer value of `mode` would change from [-1, 3] to [0, 4] - which probably is no problem, except for regression.

@MartinOtter What do you mean?

HansOlsson, 2023-10-25:
> Would a conversion script "type conversion from Integer to enumeration" work seamlessly?

I will have to investigate. 
However, as far as I understand conversion scripts wont be this time, but for version 5.

tobolar, 2023-10-26:
> However, as far as I understand conversion scripts wont be this time, but for version 5.

I understand the same way. So set the milestone to 5.0.0?

DagBruck, 2023-11-10:
> That was what you meant @DagBruck ?

@HansOlsson indeed it was.