modelica/ModelicaStandardLibrary #3409: Not fully specified initial system in Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis1

modelonrobinandersson, 2020-02-13:
In order to setup the initial equation system to Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis1 additional equations are needed in order to obtain a balanced system.

That the equations below are added is natural as a consequence of the start values and fixed=true for these states:
```
winding1.i = 0;
winding2.i = 0;
core.MagRel = 0.5;
```
However, the fourth equation that is needed is not clearly defined, and different compiler heuristics will lead to different initial equations.
We suggest to explicitly add 
```
initial equation
    core.derHstat = 0.0;
```

hubertus65, 2020-02-14:
I think that a complete specification of the initialization is necessary to achieve the same results across different tools. It would be awesome if CI could check for this in some way and flag this as a QA issue. A first step could be to make this a clearly formulated best practice. I think it is already, but I am not sure.

beutlich, 2020-03-02:
Resolved in master and maint/3.2.3 branches.

closed 2020-03-02 by beutlich