modelica/ModelicaStandardLibrary #3412: Analysis of the MSL v4.0.0-beta.1 regressions vs. v3.2.3-rc.2

beutlich, 2020-02-13:
This is the public [report](https://beutlich.github.io/MSL400BETA1FAILED/) of the 32 failed examples models for MSL v4.0.0-beta.1 vs. v3.2.2-rc.2. You can browse a preview [here](https://htmlpreview.github.io/?https://github.com/beutlich/MSL400BETA1FAILED/blob/master/index.html).

I believe many of the changes are false-positives due to updated examples. Please check, analyze and give feedback.

 - [ ] Modelica.Blocks.Examples.NoiseExamples.ActuatorWithNoise.ActuatorWithNoise
 - [ ] Modelica.Blocks.Examples.NoiseExamples.DrydenContinuousTurbulence.DrydenContinuousTurbulence
 - [ ] Modelica.Electrical.Analog.Examples.OpAmps.Differentiator.Differentiator
 - [ ] Modelica.Electrical.Analog.Examples.OpAmps.HighPass.HighPass
 - [ ] Modelica.Electrical.Analog.Examples.OpAmps.Integrator.Integrator
 - [ ] Modelica.Electrical.Analog.Examples.OpAmps.LowPass.LowPass
 - [ ] Modelica.Electrical.Machines.Examples.InductionMachines.IMC_InverterDrive.IMC_InverterDrive
 - [ ] Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_R.ThyristorCenterTap2mPulse_R
 - [ ] Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_RL.ThyristorCenterTap2mPulse_RL
 - [ ] Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_RLV.ThyristorCenterTap2mPulse_RLV
 - [ ] Modelica.Electrical.PowerConverters.Examples.DCDC.ChopperStepDown.ChopperStepDown_R.ChopperStepDown_R
 - [ ] Modelica.Electrical.PowerConverters.Examples.DCDC.ChopperStepDown.ChopperStepDown_RL.ChopperStepDown_RL
 - [ ] Modelica.Electrical.PowerConverters.Examples.DCDC.ChopperStepUp.ChopperStepUp_R.ChopperStepUp_R
 - [ ] Modelica.Fluid.Examples.AST_BatchPlant.BatchPlant_StandardWater.BatchPlant_StandardWater
 - [ ] Modelica.Fluid.Examples.HeatingSystem.HeatingSystem
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.ComparisonPolyphase.IMS_Start_Polyphase.IMS_Start_Polyphase
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_DOL.IMC_DOL
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Initialize.IMC_Initialize
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Steinmetz.IMC_Steinmetz
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Transformer.IMC_Transformer
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_withLosses.IMC_withLosses
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_YD.IMC_YD
 - [ ] Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMS_Start.IMS_Start
 - [ ] Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Conveyor.IMC_Conveyor
 - [ ] Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Initialize.IMC_Initialize
 - [ ] Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Transformer.IMC_Transformer
 - [ ] Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_YD.IMC_YD
 - [ ] Modelica.Mechanics.MultiBody.Examples.Elementary.PointGravityWithPointMasses2.PointGravityWithPointMasses2
 - [ ] Modelica.Mechanics.MultiBody.Examples.Loops.Engine1b.Engine1b
 - [ ] Modelica.Mechanics.MultiBody.Examples.Loops.Engine1b_analytic.Engine1b_analytic
 - [ ] Modelica.StateGraph.Examples.FirstExample_Variant2.FirstExample_Variant2
 - [ ] Modelica.StateGraph.Examples.FirstExample_Variant3.FirstExample_Variant3

beutlich, 2020-02-13:
The two StateGraph models FirstExample_Variant2 and FirstExample_Variant3 are false-positives due to an increased simulation time applied by #3386.

tobolar, 2020-02-14:
- Modelica.Mechanics.MultiBody.Examples.Elementary.PointGravityWithPointMasses2.PointGravityWithPointMasses2
   - Not really a problem since relative errors are about 1E-10.
- Modelica.Mechanics.MultiBody.Examples.Loops.Engine1b.Engine1b
   - Initial conditions were changed in c119e90a6642627e4f3131ec22388f10098958c9 (e.g. `inertia.w.start = 10` in `Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.Engine1Base` now). So new results are fine.
- Modelica.Mechanics.MultiBody.Examples.Loops.Engine1b_analytic.Engine1b_analytic
   - ditto

christiankral, 2020-02-14:
Change of result in `Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.ComparisonPolyphase.IMS_Start_Polyphase.IMS_Start_Polyphase ` is not relevant, as the deviation is negligible low

christiankral, 2020-02-15:
> Change of result in `Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.InductionMachines.ComparisonPolyphase.IMS_Start_Polyphase.IMS_Start_Polyphase ` is not relevant, as the deviation is negligible low

Same issue: The error of `Modelica.Electrical.Machines.Examples.InductionMachines.IMC_InverterDrive.IMC_InverterDrive` is in the zero sequence current only and thus not relevant to the simulation, as it is numeric noise.

christiankral, 2020-02-15:
The models 

- `Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_R.ThyristorCenterTap2mPulse_R`
- `Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_RL.ThyristorCenterTap2mPulse_RL`
- `Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_RLV.ThyristorCenterTap2mPulse_RLV`

are all OK from my point of view. In each model there is only one infinitely short difference in the state of one thyristor before reaching the periodic patterns of switching. This is not a modeling issue but rather a solver issue which does not have any other impact on any other variable.

christiankral, 2020-02-15:
The models 
- `Modelica.Electrical.PowerConverters.Examples.DCDC.ChopperStepDown.ChopperStepDown_R.ChopperStepDown_R`
- `Modelica.Electrical.PowerConverters.Examples.DCDC.ChopperStepDown.ChopperStepDown_RL.ChopperStepDown_RL`
- `Modelica.Electrical.PowerConverters.Examples.DCDC.ChopperStepUp.ChopperStepUp_R.ChopperStepUp_R`

have been modified and show a thus a different behavior.

christiankral, 2020-02-15:
> Same issue: The error of `Modelica.Electrical.Machines.Examples.InductionMachines.IMC_InverterDrive.IMC_InverterDrive` is in the zero sequence current only and thus not relevant to the simulation, [...]

The same issue with zero sequence components applies to all other listed FundamentalWave machine models. This is not relevant to the machine behavior as all other states are not significantly affected.

christiankral, 2020-02-15:
@ahaumer and @MartinOtter please check the following models:

- `Modelica.Blocks.Examples.NoiseExamples.ActuatorWithNoise.ActuatorWithNoise`
- `Modelica.Blocks.Examples.NoiseExamples.DrydenContinuousTurbulence.DrydenContinuousTurbulence`

@AHaumer please check the following models as you may have modified them:

- `Modelica.Electrical.Analog.Examples.OpAmps.Differentiator.Differentiator`
- `Modelica.Electrical.Analog.Examples.OpAmps.HighPass.HighPass`
- `Modelica.Electrical.Analog.Examples.OpAmps.Integrator.Integrator`
- `Modelica.Electrical.Analog.Examples.OpAmps.LowPass.LowPass`

maltelenz, 2020-02-18:
> Same issue: The error of `Modelica.Electrical.Machines.Examples.InductionMachines.IMC_InverterDrive.IMC_InverterDrive` is in the zero sequence current only and thus not relevant to the simulation, as it is numeric noise.

In that case, should that variable be part of the comparison variables? If the differences in a variable are not relevant to the simulation, then I don't think it is a sensible comparison variable.

beutlich, 2020-02-22:
> 
> 
> @AHaumer and @MartinOtter please check the following models:
> 
>     * `Modelica.Blocks.Examples.NoiseExamples.ActuatorWithNoise.ActuatorWithNoise`
> 
>     * `Modelica.Blocks.Examples.NoiseExamples.DrydenContinuousTurbulence.DrydenContinuousTurbulence`

This is due to #3224 and can be avoided by utilizing the very same fixed local seed that was computed in MSL v3.2.3. See #3434 for a possible resolution.

beutlich, 2020-02-26:
> @AHaumer please check the following models as you may have modified them:
>
>     * `Modelica.Electrical.Analog.Examples.OpAmps.Differentiator.Differentiator`
>
>     * [...]

@AHaumer Please analyze.

casella, 2020-03-02:
Regarding `Modelica.Fluid.Examples.AST_BatchPlant.BatchPlant_StandardWater.BatchPlant_StandardWater`, in PR #2925 I changed the `dp_nominal` values of the valves by about 1000 Pa to improve the numerical conditioning of the system. As a consequence, discrepancies of up to 1000 Pa in the pressures are to be expected, and they are ok. I understand all other variables basically have the same values, which confirm that the changes in PR #2925 were good.

casella, 2020-03-02:
Regarding `Modelica.Fluid.Examples.HeatingSystem.HeatingSystem`, in #3236 I changed the medium model to a much simpler fluid model with constant cp, that is appropriate for the temperature and pressure range of the system, and leads to faster and more robust simulation. The very small change in the enthalpies is the consequence of the slightly different value of cp, and is absolutely acceptable.

casella, 2020-03-03:
@beutlich, @christiankral, @kristinmajetta, @christophclauss,  I also had a look at all the opamp example results. The opamp voltage output is completely different in all cases. 

The examples have been restructured to use inheritance, I thought the intention was to end up with the same results, but apparently this is not the case.

I focused on one specifica case, namely LowPass. I tried to [blame](https://github.com/modelica/ModelicaStandardLibrary/blame/master/Modelica/Electrical/Analog/Examples/OpAmps/LowPass.mo) it first, but unfortunately the file has been split by @christiankral 18 days ago, so it is not obvious which changes (if any) were applied to those models.

I then analyzed LowPass manually, comparing it to the MSL 3.2.3 setup. First of all, the old library had a trapezoid voltage input, while the new one has a voltage pulse input. I'm not sure why this was changed, I guess we should keep the original one.

Regarding the circuit parameters, with the old MSL 3.2.3 we have
```
f = 10; // Frequency of input signal in Hz
fG = f/10; // Low-pass cut-off frequency in Hz"
C = 1/(2*pi*fG*R2); // computation of the capacitance to get the required cut-off frequency
```
note that the low-pass frequency fG (in Hz) was correctly converted into rad/s by multiplying it times 2*pi.

If you now look at the new implementation of [FirstOrder.mo](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Electrical/Analog/Examples/OpAmps/OpAmpCircuits/FirstOrder.mo) and [LowPass.mo](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Electrical/Analog/Examples/OpAmps/LowPass.mo), you see the following parameter binding equations:
```
T = 0.1/f; // from LowPass
C = T/R2; // from FirstOrder
``` 
from which` C = T/R2 = 0.1/(f*R2)`. Considering that `f = 10*fG` in the original model, you get `C  = 0.1/(10*fG*R2) = 1/(100*fG*R2)`, while in the original model you got C = 1/(2*pi*fG*R2). 

It is apparent that the new implementation has a value of C which is 100/pi = 31 times smaller, so the low-pass filter is not really filtering that much of the original signal. Of course the value one gives to T in these examples is somwhat arbitrary, but while I understood the rationale of the original example, i.e. having a cut-off frequency at one-tenth of the signal base frequency, I don't really understand the rationale of the restructured model.

This specific example could be fixed by changing the binding equation in LowPass to
```
T = 1/(2*pi*0.1*f);
```
which corresponds to the rationale of the original example, and by restoring the original signal source to TrapezoidVoltage instead of PulseVoltage.

I guess similar analysis apply to the other test cases. 

@christiankral can you comment on this? I can help restoring the original model outputs with the new structure if you don't have time now.

christiankral, 2020-03-03:
> @beutlich, @christiankral, @kristinmajetta, @christophclauss, @AHaumer, I also had a look at all the opamp example results. The opamp voltage output is completely different in all cases.

I guess @ahaumer shall comment on this issue, as he updated the models

Refs https://trac.openmodelica.org/OpenModelica/ticket/5857

christiankral, 2020-03-03:
> I focused on one specifica case, namely LowPass. I tried to blame it first, but unfortunately the file has been split by @christiankral 18 days ago, so it is not obvious which changes (if any) were applied to those models.
>
>I then analyzed LowPass manually, comparing it to the MSL 3.2.3 setup. First of all, the old library had a trapezoid voltage input, while the new one has a voltage pulse input. I'm not sure why this was changed, I guess we should keep the original one.
>
> ...
>
> @christiankral, @AHaumer, can you comment on this? I can help restoring the original model outputs with the new structure if you don't have time now.

Same here @AHaumer shall comment on this issue, as he changed the models

AHaumer, 2020-03-07:
I have to admit: It's hard to remember why, but I've changed the 4 OpAmp-examples. 
There are 2 optons:

1. Just accept the changes of the examples and define new reference results.
2. I prepared a PR where these 4 OpAmp-examples have been changed to produce the same results as the old examples.

beutlich, 2020-03-07:
OpAmp changes have been introduced by #2979. If we lack a proper motivation, I am in favour of option 2.

closed 2020-03-07 by beutlich