modelica/ModelicaStandardLibrary #3415: Analysis of failing test models of ModelicaTest v4.0.0-beta.1

beutlich, 2020-02-13:
The following test models do fail during check, translation or simulation (with tolerance=1e-06).

Please analyze and provide feedback.

 - [ ] ModelicaTest.Magnetic.FluxTubes.BasicComponents
```
Error: The following parameters don't have any value, only a start value:
    leakageWithCoefficient.c_usefulFlux(start = 0.7)
```

- [ ] ModelicaTest.MultiBody.Forces.EngineGasForce
```
Error: The variable position.v has fixed = true, but it was deduced to be constant: 0
```

 - [ ] ModelicaTest.Fluid.TestComponents.Sensors.TestTemperature1
```
Error: The following error was detected at time: 6.991863958612594
  OnePortTemperature and TwoPortTemperature shall give the same result
  Failed condition: abs(Tdiff)/50 < 0.001
```

 - [ ] ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse
```
ERROR: Failed to solve nonlinear system using Newton solver.
  Time: 1.768
```

 - [ ] ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank
```
Error: The following error was detected at time: 10.00000000249371
  
Temperature T (= -15514.7 K) is not
in the allowed range (272.15 K <= T <= 403.15 K)
required from medium model "SimpleLiquidWater".

  Failed condition: tank.medium.T >= 272.15 and tank.medium.T <= 403.15
```

 - [ ] ModelicaTest.Fluid.TestPipesAndValves.DynamicPipeClosingValve
```
Simulation hangs and needs o be manually aborted.
```

 - [ ] ModelicaTest.Media.TestAllProperties.MoistAir
```
Error: The following error was detected at time: 0
  Array dimensions did not match
  Failed condition: RealMatchingSizes(a,b)

The stack of functions is:
Modelica.Media.Air.MoistAir.s_pTX
Modelica.Media.Air.MoistAir.T_psX.f_nonlinear
Modelica.Math.Nonlinear.solveOneNonlinearEquation
Modelica.Math.Nonlinear.solveOneNonlinearEquation(
function(Modelica.Media.Air.MoistAir.T_psX.f_nonlinear(, 202650, Modelica.Media.Air.MoistAir.specificEntropy(
state), state.X[1:1])), 
190, 
647, 
1E-13)
Error: Integrator failed to start model.
```

 - [ ] ModelicaTest.Utilities.TestStrings
```
Error: The following error was detected at time: 0
  Strings.hashString 1 failed
  Failed condition: hash1 == 1827717433
```

HansOlsson, 2020-02-14:
@tobolar ModelicaTest.MultiBody.Forces.EngineGasForce was introduced when the test was introduced in 
https://github.com/modelica/ModelicaStandardLibrary/commit/b47f08544bf782eb522563156c754af124976cf2
and I believe the fixed=true should just be removed.

HansOlsson, 2020-02-14:
ModelicaTest.Fluid.TestComponents.Sensors.TestTemperature1 just has a slightly too sensitive test.

ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank is a genuine problem when the tank is empty and starts filling up.

ModelicaTest.Media.TestAllProperties.MoistAir is related to an old issue with the interface of Modelica.Media.Air.MoistAir.s_pTX - which takes `X[:]`, but needs `X[nX]`, and for some reason `Modelica.Media.Air.MoistAir.T_psX` uses `X[1:nXi]` - which will not work. The simple solution would be to use plain `X`, or `X[1:nX]`.

It was introduced by https://github.com/beutlich/ModelicaStandardLibrary/commit/3a5991fb886401024162eda5f4d6ae817df26a36 and seems like trivial typo, one function used `X[1:nXi]` and the next `X[1:nX]`; and both were changed to `X[1:nXi]`.

tobolar, 2020-02-14:
@HansOlsson 
> ...and I believe the fixed=true should just be removed.

Confirmed.
We could then add "annotation(Dialog(enable=not exact))" to `v`and `a` in `Modelica.Mechanics.Translational.Sources.Position`. Possibly also "showStartAttribute = not exact"?

HansOlsson, 2020-02-14:
> 
> 
> @HansOlsson
> 
> > ...and I believe the fixed=true should just be removed.
> 
> Confirmed.
> We could then add "annotation(Dialog(enable=not exact))" to `v`and `a` in `Modelica.Mechanics.Translational.Sources.Position`. 

Yes, good idea to avoid this problem in the future.

> Possibly also "showStartAttribute = not exact"?

Preferably not. At least Dymola does not handle non-literal values for this one at the moment, and it seems a bit messy in terms of user-experience.

beutlich, 2020-02-14:
@HansOlsson ModelicaTest.Utilities.TestStrings looks like a Dymola issue to me since both SimulationX and WSM report the expected hash values. Does Dymola 2020 use a different hash algorithm than the default one?

beutlich, 2020-02-14:
> ModelicaTest.Media.TestAllProperties.MoistAir is related to an old issue with the interface of Modelica.Media.Air.MoistAir.s_pTX - which takes `X[:]`, but needs `X[nX]`, and for some reason [...]

Thanks for the analysis. I had a look myself but was not able to see it instantly.

> It was introduced by [beutlich@3a5991f](https://github.com/beutlich/ModelicaStandardLibrary/commit/3a5991fb886401024162eda5f4d6ae817df26a36) and seems like trivial typo, one function used `X[1:nXi]` and the next `X[1:nX]`; and both were changed to `X[1:nXi]`.

Luckily I also created that test model by c0c48961e7429b3f371317a82cbc76bb009ab79a, which once again shows the importance of unit tests.

christiankral, 2020-02-14:
Fix of ModelicaTest.Magnetic.FluxTubes.BasicComponents is provided by #3424

beutlich, 2020-02-22:
> @HansOlsson ModelicaTest.Utilities.TestStrings looks like a Dymola issue to me since both SimulationX and WSM report the expected hash values. Does Dymola 2020 use a different hash algorithm than the default one?

@HansOlsson Can you please cross-check?

HansOlsson, 2020-02-24:
> 
> 
> > @HansOlsson ModelicaTest.Utilities.TestStrings looks like a Dymola issue to me since both SimulationX and WSM report the expected hash values. Does Dymola 2020 use a different hash algorithm than the default one?
> 
> @HansOlsson Can you please cross-check?

Hmm... I'm trying to check, but there is something confusing:

```
#define HASH_AP(key, keylen, hash) ...
int ModelicaStrings_hashString(_In_z_ const char* str) {
    ...
    HASH_VALUE(str, len, h.iu);
```
Why define HASH_AP and then use HASH_VALUE (and it seems HASH_VALUE depends on uthash.h; which may use different algorithms).

beutlich, 2020-02-24:
Yes, this this the whole idea:

https://github.com/modelica/ModelicaStandardLibrary/blob/69d20512b37dc304936852ca1df7f61e38e9274b/Modelica/Resources/C-Sources/ModelicaStrings.c#L90-L91

HansOlsson, 2020-02-24:
> 
> 
> Yes, this is the whole idea:
> 
> https://github.com/modelica/ModelicaStandardLibrary/blob/69d20512b37dc304936852ca1df7f61e38e9274b/Modelica/Resources/C-Sources/ModelicaStrings.c#L90-L91

Hmm.... So there are two possible issues:

- Some difference in uthash.h
- Dymola combine several C-files when compiling, and some other file include uthash.h first (the normal multiple inclusion guard will then make the new definition of HASH_FUNCTION meaningless).

HansOlsson, 2020-02-24:
> Hmm.... So there are two possible issues:
>
>     * Some difference in uthash.h
>
>     * Dymola combine several C-files when compiling, and some other file include uthash.h first (the normal multiple [...]

It's the latter. I would thus propose the following:
Don't include uthash.h in ModelicaStrings.c, and replace call of HASH_VALUE by HASH_AP.

I don't see how that can cause problems for other tools.

beutlich, 2020-02-24:
Can you please try. In ModelicaStrings.c, instead of 

```c
#define HASH_FUNCTION HASH_AP 
#include "uthash.h"
```

have

```c
#include "uthash.h"
#undef HASH_FCN
#define HASH_FCN HASH_AP
```

beutlich, 2020-02-24:
Otherwise, I'd rather consider it as a tool-issue.

HansOlsson, 2020-02-24:
> 
> 
> Can you please try. In ModelicaStrings.c, instead of
> 
> ```c
> #define HASH_FUNCTION HASH_AP 
> #include "uthash.h"
> ```
> 
> have
> 
> ```c
> #include "uthash.h"
> #undef HASH_FCN
> #define HASH_FCN HASH_AP
> ```

In testing that works as well, but I find it fragile (in similar scenarios with multiply included C-files). The reason is that we are using macros with multiple definitions in the sources, and I don't see what we try to achieve.

I would thus prefer to just remove uthash.h from ModelicaStrings.c (i.e. reverting #2262 ).

Or in more detail I see the following possibilities:

- Remove uthash.h from ModelicaStrings.c; and just use HASH_AP directly.
- Make ModelicaStrings.c use what-ever hash-function we define in uthash.h (i.e. remove HASH_AP) and remove the regression-test for the value of hashString.

beutlich, 2020-02-24:
> Make ModelicaStrings.c use what-ever hash-function we define in uthash.h (i.e. remove HASH_AP) and remove the regression-test for the value of hashString.

Yes, sounds possible. `HASH_JEN` will be the "new" hash function then.

HansOlsson, 2020-02-24:
> 
> 
> > Make ModelicaStrings.c use what-ever hash-function we define in uthash.h (i.e. remove HASH_AP) and remove the regression-test for the value of hashString.
> 
> Yes, sounds possible. `HASH_JEN` will be the "new" hash function then.

I'll make the PR for this change then.

beutlich, 2020-02-24:
ModelicaTest.Fluid.TestPipesAndValves.DynamicPipeClosingValve simulates successfully if the tolerance is increased to e,g,. 2e-5.

beutlich, 2020-02-25:
ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse successfully simulates if the output interval is increased again.

```
simulateModel("ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse", stopTime=4, method="Dassl", tolerance=1e-6, resultFile="TestValvesCompressibleReverse");
 = true
simulateModel("ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse", stopTime=4, method="Dassl", tolerance=1e-6, resultFile="TestValvesCompressibleReverse", outputInterval=0.0008);
 = false
simulateModel("ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse", stopTime=4, method="Dassl", tolerance=1e-6, resultFile="TestValvesCompressibleReverse", outputInterval=0.008);
 = true
simulateModel("ModelicaTest.Fluid.TestComponents.Valves.TestValvesCompressibleReverse", stopTime=4, method="Dassl", tolerance=1e-4, resultFile="TestValvesCompressibleReverse", outputInterval=0.0008);
 = true
```

Not sure, if this is a tool issue or not. But according to https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v3.2.3/ModelicaTest/Fluid/TestComponents/Valves/TestValvesCompressibleReverse/creation.txt#L26 it is not a regression issue.

beutlich, 2020-02-25:
Neither is ModelicaTest.Fluid.TestComponents.Vessels.TestSimpleTank an actual regression, see https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v3.2.3/ModelicaTest/Fluid/TestComponents/Vessels/TestSimpleTank/creation.txt#L26.

closed 2020-02-25 by beutlich