modelica/ModelicaStandardLibrary #3420: Analysis of the ModelicaTest v4.0.0-beta.1 regressions vs. v3.2.3-rc.2

beutlich, 2020-02-14:
This is the public [report](https://beutlich.github.io/MSLTEST400BETA1FAILED/index.html) of the 6 failed examples models for MSL v4.0.0-beta.1 vs. v3.2.2-rc.2.

Please check, analyze and give feedback.

- [ ] ModelicaTest.Fluid.TestExamplesVariants.HeatingSystem_FixedInitial.HeatingSystem_FixedInitial
- [ ] ModelicaTest.Fluid.TestExamplesVariants.HeatingSystem_SteadyState.HeatingSystem_SteadyState
- [ ] ModelicaTest.Media.TestAllProperties.FlueGasSixComponents.FlueGasSixComponents
- [ ] ModelicaTest.Media.TestAllProperties.SimpleAir.SimpleAir
- [ ] ModelicaTest.MultiBody.InitializationConversion.Joints.Joints
- [ ] ModelicaTest.MultiBody.Joints.UniversalSpherical.UniversalSpherical

tobolar, 2020-02-14:
- ModelicaTest.MultiBody.InitializationConversion.Joints.Joints
   - Difference due to deletion of `s_offset` from `MultiBody.Joints.Prismatic` joint. Previously: prismatic.s_offset=0.1

beutlich, 2020-02-14:
Shall we restore it?

tobolar, 2020-02-14:
An option would be to convert s_offset to s(start = s.start<previousversion> + s_offset) but I don't know if this is possible. And a question is whether to also set `s.fixed=true` in order to retain the influence of `s_offset` on `s` then.

HansOlsson, 2020-02-14:
> 
> 
> An option would be to convert s_offset to s(start = s.start + s_offset) but I don't know if this is possible. And a question is whether to also set `s.fixed=true` in order to retain the influence of `s_offset` on `s` then.

That seems a bit too complicated:

It seems that you could set s_offset for a component, and then hierarchically modify s.start, and I don't see how we could update that hierarchical modifier for s.start in a straightforward way.

tobolar, 2020-02-19:
Then, probably best to really restore `s_offset` as there is no possibility for a feasible conversion. See also #2465.

tobolar, 2020-02-19:
ModelicaTest.MultiBody.Joints.UniversalSpherical: significant deviations only for system branch containing `bodyBox1`. It is caused by changed direction of rotation axes of joint `universal`:
- now:n_a={0,0,1}, n_b={0,1,0}
- previously: n_a={1,0,0}, n_b={0,1,0}

Also, initialization has changed.
So the results are fine.

tobolar, 2020-02-21:
To removement of `s_offset`:
1. In 68e893f5d163395bad946b24fa00fa156dc24468 the two original joints (Prismatic and Revolute) were copied into `ObsoleteModelica4.Mechanics.MultiBody.Joints`. So the conversion script shall be adapted accordingly to convert these elements into those from ObsoleteModelica4. At the moment, it only "eliminates" `s_offset`, commit b8a0658131936825df76a6f891c18366c4965542.
2. A conversion such as `convertClassIf("OldClassName","para","val","NewClassName");` which only *optionally* convert components seems reasonable.
3. There shall be documented how user shall proceed to change from obsolete joints to proper ones. Where is the best place for such a description?

beutlich, 2020-02-22:
The previous discussion about removal of constants s_offset and phi_offset happened in #3048. I also asked if it is legal if a constant is modified and if we should adapt ModelicaTest.MultiBody to get reproducible results. Obviously we failed for one model. (It even looks like an earlier default when @MartinOtter changed the variability from parameter to constant.)

I see three possibilities:
1. We reintroduce s_offset and phi_offset, but as proper (Dialog) parameters (as still present in the assembly joints).
1. We touch ModelicaTest.MultiBody.InitializationConversion.Joints.Joints to give identical results.
1. We do not touch ModelicaTest.MultiBody.InitializationConversion.Joints.Joints and accept the modified results.

Whatever @tobolar's and @MartinOtter's preference is, we need to adapt it immediately.

hubertus65, 2020-02-26:
To the best of my analysis, the errors in Media are spurious errors in the csv-compare code. 

When I open the detailed error reports, I find: 

FlueGasSixComponents.err_h_is
5003 (relative error is 9,16569193376174E-10)
SimpleAir.s_is
5003 (relative error is 5,48431446405923E-12)
5003 (relative error is 5,48431446405923E-12)

Which is extremely small relative errors, that trigger an error report even though the tolerance is set to 0.002.  in the first case, similar the second. 

If it helps automatic testing, use the new results as golden results, but the issue here is really in csv-compare (I think), so this could come up again. All shoudl be ok then for Modelica.Media

beutlich, 2020-02-29:
> 3. We do not touch ModelicaTest.MultiBody.InitializationConversion.Joints.Joints and accept the modified results.

Accepting option 3.

beutlich, 2020-02-29:
The two HeatingSystem test models extend from Modelica.Fluid.Examples.HeatingSystem where the medium Modelica.Media.Water.StandardWater was updated to Modelica.Media.CompressibleLiquids.LinearWater_pT_Ambient by #3260.

closed 2020-02-29 by beutlich

tobolar, 2020-03-06:
@beutlich , @MartinOtter 
> Accepting option 3.

Independent of this, there are stil unresolved questions in https://github.com/modelica/ModelicaStandardLibrary/issues/3420#issuecomment-589566805.

So especially: 
1. How to convert?
   1. Ignore s_offset, as does the current conversion or
   2. to convert to obsolete classes?
2. How/where to inform a user about this.

beutlich, 2020-03-06:
I can only repeat myself:

> The previous discussion about removal of constants s_offset and phi_offset happened in #3048. I also asked if it is legal if a constant is modified and if we should adapt ModelicaTest.MultiBody to [...]

MartinOtter, 2020-04-06:
As I recall, I wanted to make the definition of the joints simpler and at one point in time (probably 2007) changed parameter to constant so that this data is no longer visible in parameter menus. Dymola supported at that time to modify constants (during compilation; not during run-time). This is probably Modelica semantics because this feature is heavily used in Modelica.Media. 

I am unsure what to do. Automatic translation seems to be overkill. Maybe it is o.k. to just remove them because they are hidden since such a long time (and only mention it in the documentation). If there is the unlikely case that someone used them, there will be an error message (so will not be silently wrong).