modelica/ModelicaStandardLibrary #3425: Proposal to split Modelica.Blocks into single files

christiankral, 2020-02-15:
I am asking the library officers @AHaumer and @MartinOtter if they have any intentions to have the package `Modelica.Blocks` split into single files. In case you agree, it makes sense to have this done before MSL 4.0.0 gets released.

AHaumer, 2020-02-15:
It makes sense, but I'd ask for @MartinOtter 's opinion.

christiankral, 2020-03-01:
@MartinOtter If you agree, I can prepare a PR to get the package split into individual files. On my opinion the great advantage of single files is, that a GitHub history can directly be created for each Modelica class of the package. This way it is very easy to check what changes have been applied to each Modelica class.

beutlich, 2020-03-02:
Since there is an open PRs for "Blocks" I am removing the milestone.

MartinOtter, 2020-03-03:
> @MartinOtter If you agree, I can prepare a PR to get the package split into individual files. On my opinion the great advantage of single files is, that a GitHub history can directly be created for [...]

This is fine from my side